cantools/cantools #113: DBC parser fails if "SG_MUL_VAL_" not in "NS_" list

HubertD, 2019-02-28:
Hello,

I've got a bunch of DBC files from a customer that all throw syntax errors at the first message definition when trying to import via cantools.database.load_file("xyz.dbc").

Did not try to understand the parser code, but turns out that in the "NS_" section, said DBCs lack the "SG_MUL_VAL_" property that the cantools test DBCs have.
Adding this property in my DBCs, the parser seems to work fine.

Problem can also be triggered when removing the "SG_MUL_VAL_" line from other files, e.g. cantools/tests/files/dbc/foobar.dbc : [foobar.dbc.txt](https://github.com/eerimoq/cantools/files/2915065/foobar.dbc.txt)

Thanks for your great work,

Hubert

eerimoq, 2019-02-28:
Fixed in version 32.11.2, available on PyPI.

HubertD, 2019-03-01:
Confirming this is fixed. v32.11.2 works for my DBCs :1st_place_medal:

closed 2019-03-01 by HubertD