cantools/cantools #115: UnsupportedDatabaseFormatError occurs when parsing arxml file

BrushLee90, 2019-03-04:
After upgrading to the latest release version, there is an index error occurring as below：
[image]
The using test arxml file is listed:
[test_GEE.zip](https://github.com/eerimoq/cantools/files/2924972/test_GEE.zip)

BrushLee90, 2019-03-04:
in case you can not see the error snipping,  I copy the error text as following:
Exception has occurred: cantools.database.UnsupportedDatabaseFormatError
ARXML: "SW-BASE-TYPE at /BaseTypes/Perc4_2 does not exist."
  File "D:\cantestpython\d\cantools\database\__init__.py", line 336, in load_string
    raise UnsupportedDatabaseFormatError(e_arxml, e_dbc, e_kcd, e_sym, e_cdd)
  File "D:\cantestpython\d\cantools\database\__init__.py", line 247, in load
    strict)
  File "D:\cantestpython\d\cantools\database\__init__.py", line 179, in load_file
    strict)
  File "D:\cantestpython\d\temprun.py", line 13, in <module>
    dbcbf2 = cantools.database.load_file("test_GEE.arxml","arxml")

eerimoq, 2019-03-04:
With the latest version (32.11.2) the error is ``error: ARXML: "SW-BASE-TYPE at /BaseTypes/Perc4_2 does not exist."``, not yours. The ARXML-file is probably incomplete, the BaseTypes package is missing.

closed 2019-03-06 by eerimoq