cantools/cantools #116: Read signal Initial value from dbc file not works

BrushLee90, 2019-03-04:
I found that it can't read signal initial value from DBC file. 
I use a test dbc file (has uploaded as an attachment ), in which the signal "MRRSt" value in message 0x3a4 is set to 6. However, the read value is "None". 

[image]

[image]
[dbc_tes_file.zip](https://github.com/eerimoq/cantools/files/2925505/dbc_tes_file.zip)

eerimoq, 2019-03-04:
The initial value is not yet extracted from any database format. Feel free to implement support for it for your particular format.

eerimoq, 2019-03-06:
Try release 32.12.0, available on PyPI. It has intial support for inital values in DBC-files.

BrushLee90, 2019-03-07:
Wonderful! Thank you for your work~~

closed 2019-03-07 by BrushLee90