cantools/cantools #117: How to read from asc file ?

dglent, 2019-03-05:
Hi,

i'd like to decode frames from an .asc file:
   0.0118	2	208	Rx	d	8 18 B8 32 00 4C 32 32 32

I tried with db.decode_message but i don't know how to encode correctly the message '18 B8 32 00 4C 32 32 32' to send and i obtain the error : "bitstruct.Error: unpack requires at least 72 bits to unpack (got 64)"

what would be the right way to decode it ?

Thanks in advance

eerimoq, 2019-03-05:
I don't know how to interpret the ASC file format. Sorry.

karlding, 2019-03-05:
The way I usually work with ASC files is to:

1. Install `can-utils` userspace applications
1. Convert the ASC to a log file using `asc2log`
1. Bring up a SocketCAN interface
1. Use `canplayer` and replay the log file
1. Use `cantools` to decode data over SocketCAN

Assuming you're using a Debian-based system (like Ubuntu), it'll look something like this:

```bash
# Install can-utils
sudo apt-get install can-utils

# Convert ASC to log file
asc2log -I 2019-01-09_18-34-16_844.asc -O 2019-01-09_18-34-16_844.log

# Load kernel modules
sudo modprobe can
sudo modprobe can_raw
sudo modprobe vcan

# Bring up SocketCAN interfaces
sudo ip link add dev vcan0 type vcan
sudo ip link set up vcan0

# Replay data logged on can0 over vcan0
canplayer -I 2019-01-09_18-34-16_844.log vcan0=can0

# Decode using cantools
cantools monitor --single-line can_network.dbc
```

dglent, 2019-03-07:
Thanks i will take this as an example in order to convert asc>log in python

closed 2019-03-11 by eerimoq

schorschie, 2021-04-06:
Is there an easier way, which also works on windows? Decoding an ASCII file doesn't seem to be such an out of the ordinary use case?

longnt0707, 2021-07-28:
> Is there an easier way, which also works on windows? Decoding an ASCII file doesn't seem to be such an out of the ordinary use case?

Succeed in decoding ASC on Windows as following:

```
# Load dbc file to database
db = cantools.database.Database()
with open (dbc_file_name, 'r') as dbcfile:
        db.add_dbc(dbcfile)

# Use ASCReader from python-can (installed as dependency for cantools)
with can.ASCReader(asc_file_name) as asclog:
    for mesg in asclog:
        dced_msg = db.decode_message(mesg.arbitration_id, mesg.data)
```