cantools/cantools #122: does not support tp messsage

sgnes, 2019-03-12:
hello,

after using this package, I found this does not support J1939TP message, like if a message is longer than 8 byte, we need to use the J1939TP to send it.
Can you please confirm this point?
Thanks.

eerimoq, 2019-03-12:
I don't know what J1939TP means, so you are probably right.

dudenamedben, 2019-03-15:
I have also encountered problems with this when using cantools, as messages longer than 8 bytes were detected as invalid, which is true for a single frame (without CanFD).

However, in an abstract sens thee J1939 transport protocol defines a way to send and receive longer pieces of data using multiple packets.

> They use two special messages to facilitate these multi-packet transfers.  The first is known as a Transport Protocol Connection Management message (TP.CM).  The data in a TP.CM message contains connection commands (also called Control Byte), the PGN identifier of the multi-packet message and information about how to reconstruct the message.  The second is called a Transport Protocol Data Transfer message (TP.DT).  The data of a TP.DT message contains a sequence number in the first byte and uses the remaining seven bytes for the data of the multi-packet message.

In regards to databases, I'm not entirely sure how to represent this data.  For fixed length > 8 bytes I image you just use the TP arbitration ID with PGN, and then define the aggregated data contents.  For variable length data transfers, such as strings, maybe there is something you can do with multiplexing?  I'm interested if anybody else has feedback or references on representing this in a CAN database.

The transport protocol and bam message section can help explain this further here.
http://www.ni.com/example/31215/en/

sgnes, 2019-03-16:
https://github.com/benkfra/j1939
This repo has a tp implementation for j1939,maybe we can use this one.

msalau, 2019-03-28:
Hi,

I don't know how other tools handle this, but I've seen a dbc file that defines a message 200 byte long. May be this is it...

See message 2633832958 in https://hackage.haskell.org/package/ecu-0.0.8/src/src/j1939_utf8.dbc

Hope this helps.

mr-sour, 2019-05-11:
A long time ago in a Galaxy far away python-can had a implementation of this but it was deprecated and removed leaving it upto higher layer packages to deal with. 

So pretty much what would need to happen here for this to work is message.py would need to pick up on if the message is a TP message and I would recommend using the continuation of the original j1939 support from python-can https://github.com/milhead2/python-j1939 to send and recv these messages. I think all we would need to start is just adding the ability to allow the user to encode and decode j1939 messages greater then 8 bytes but explicitly state that you need to use python-j1939 to get and send these messages. That package should provide a message that looks like [29bit-Frame] [byte-0] ... [byte-n] and be able to send such frames, so hopefully nobody has to get there hands dirty in this repo with the specifics of j1939-TP. 

The only thing I can think of that wouldn't be handled by that library is TP messages can be fixed length or variable so if it is a fixed length message message.py would have to force the user to fill the signal data out as it does now just for larger messages. If its variable it needs to just drop the data to the user and let them handle it or have some fancy support where a user could define an encoder/decoder and attach that to the signal entry.

@msalau example is a fixed length message. I have included VIN as an example of a variable length 
Below. The VIN dbc entry it just omits pretty much omits the whole `:` and everything after it and leaves it upto the user or dbc framework to format the databytes.
~~~~
CM_ BO_ 2566843392 "Vehicle Identification";
CM_ SG_ 2566843392 VehicleIdentificationNumber "Vehicle Identification Number"; 
~~~~

kossnikita, 2024-06-25:
Is there support for J1939 TP now? Are there any implementation examples?