cantools/cantools #123: Enable "import cantools" for python installations without curses

dudenamedben, 2019-03-15:
I'm using a python build that doesn't include curses by default. Based on this contraint, I think it would be a good idea to try/catch import error for curses, as only the monitor tool uses it.

Or perhaps make the import conditional, to only import if the cantools monitor command is actually executed, so all other functionality functions as normal.

```
>>> import cantools
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/lib/python3.5/site-packages/cantools/__init__.py", line 7, in <module>
    from .subparsers import monitor as _monitor
  File "/usr/lib/python3.5/site-packages/cantools/subparsers/monitor.py", line 4, in <module>
    import curses
  File "/usr/lib/python3.5/curses/__init__.py", line 13, in <module>
    from _curses import *
ImportError: No module named '_curses'
```

eerimoq, 2019-03-15:
Fixed in version 32.13.0!

dudenamedben, 2019-03-15:
Looks good.  Thanks for the quick fix!

closed 2019-03-15 by dudenamedben

eerimoq, 2019-03-15:
No problem. Easy fix =)