cantools/cantools #124: Start bit should LSB for both little-endian and big-endian

BayarjavkhlanBatbaatar, 2019-03-25:
Hello, 

The description for the start bit of DBC file is not correct when it is big-endian signal. LSB is considered as the startbit for DBC.  [In this line](https://github.com/eerimoq/cantools/blob/5009911c404259c24a00d989fd10da0f578be6ed/cantools/database/can/signal.py#L85), startbit should be 14, not 2. On the other hand, the value 0 should be considered as a startbit, not 4. Can you please fix this issue. Thanks, :smile:  
[image]

eerimoq, 2019-03-25:
Well, the description in signal.py is the cantools internal format, not the DBC format. It might not be the most intuitive format, but I guess we'll have to live with it. You can make the change in a PR and if it turns out well I can merge it.

BayarjavkhlanBatbaatar, 2019-03-25:
@eerimoq okay. I will make a pull request.

BayarjavkhlanBatbaatar, 2019-03-25:
Hey @eerimoq, I think it is your right. Once we load a CAN configuration file, it is not any of the specific format. I think it is better to deal with it locally. I am closing the issue. Thank you. This is just a great library to use. Thank you :smile:

closed 2019-03-25 by BayarjavkhlanBatbaatar