cantools/cantools #127: cantools.database.can.bus.py only used for .kcd parsing (need .dbc support)

anxuae, 2019-03-28:
Currently the database created from a .dbc file is always created sith an empty buses list.
See https://github.com/eerimoq/cantools/blob/4bde521fb35787776092d6ec31dc1356e4b3c35a/cantools/database/can/formats/dbc.py#L1352-L1356

However, information about the bus can be found in the DBC.  For instance in the one I'm using:

*Bus name:*

```
BA_DEF_  "DBName" STRING ;
BA_DEF_DEF_  "DBName" "";
BA_ "DBName" "Infotainment_AmplifierCAN";
```

*Bus baudrate:*

```
BA_DEF_  "Bauderate" INT 0 1000000;
BA_DEF_DEF_  "Bauderate" 125000;
BA_ "Bauderate" 125000;
```

Is it possible to add at least a `Bus()` in the list with the following info?

eerimoq, 2019-03-28:
Feel free to implement it. I'll likely merge it =)

anxuae, 2019-04-04:
Hi eerimoq,
Do you think the my pull request can be integrated?
Thanks
Anxuae

eerimoq, 2019-04-04:
Sorry, I've been working on another project lately and forgot about your PR. I'll have a look now.

eerimoq, 2019-04-04:
I added tests and refactored a bit.

There was a comment that the default baudrate was copied from kcd.py. I changed the default baudrate to None, or unavailable. Let me know if it should be 500000 instead and I'll change it back.

anxuae, 2019-04-05:
Thanks a lot.
No matter of the default value, I did it only to be consistent with KCD parser.

closed 2019-04-06 by eerimoq