cantools/cantools #128: Dump "IEEE Float" to DBC not supported?

qc17, 2019-03-28:
Hi,
I have a signal:
`signal1 = cantools.database.can.Signal(sigName1, 0, 32,comment=comment1, receivers=receiver, is_float=True)`
that is added to a message/DBC. After dumping the database to a file the whole section with
`SIG_VALTYPE_ <can-id> <signame> : 1;`
is missing.

Could you please add this part in case the signal type is float or double?

Thank you for any help.

eerimoq, 2019-03-28:
Feel free to add it yourself.

Thanks!

karlding, 2019-03-30:
I ran into this issue as well, and have a patch ready to go. @qc17 are you still working on this? Otherwise, I'll just submit mine.

qc17, 2019-04-01:
I only have a quick fix to dump IEEE-float signals right now. So you are welcome to submit your patch and then I will check my further need to dump doubles to a DBC.

eerimoq, 2019-04-08:
Part of version 32.16.0, available on PyPI!

closed 2019-04-08 by eerimoq