cantools/cantools #135: Frame ID is truncated to lower two bytes when generating c source

yangliu28, 2019-04-11:
For some specific large frame ID, when generating the c source code, the bytes beyond lower two bytes will be truncated.

To repeat this error, use frame ID such as `2147490681` or `2147490681` in *tests/files/dbc/motohawk.dbc* and generate c source. Both frame IDs start with `0x8000` in the high two bytes, so it's likely due to the use of `long` instead of `unsigned long` to represent frame ID.

```
VERSION "1.0"

NS_ : 
	NS_DESC_
	CM_
	BA_DEF_
	BA_
	VAL_
	CAT_DEF_
	CAT_
	FILTER
	BA_DEF_DEF_
	EV_DATA_
	ENVVAR_DATA_
	SGTYPE_
	SGTYPE_VAL_
	BA_DEF_SGTYPE_
	BA_SGTYPE_
	SIG_TYPE_REF_
	VAL_TABLE_
	SIG_GROUP_
	SIG_VALTYPE_
	SIGTYPE_VALTYPE_
	BO_TX_BU_
	BA_DEF_REL_
	BA_REL_
	BA_DEF_DEF_REL_
	BU_SG_REL_
	BU_EV_REL_
	BU_BO_REL_
	SG_MUL_VAL_

BS_:

BU_: PCM1 FOO

BO_ 2147490681 ExampleMessage: 8 PCM1
 SG_ Temperature : 0|12@0- (0.01,250) [229.52|270.47] "degK"  PCM1,FOO
 SG_ AverageRadius : 6|6@0+ (0.1,0) [0|5] "m" Vector__XXX
 SG_ Enable : 7|1@0+ (1,0) [0|0] "-" Vector__XXX

CM_ BO_ 2147490681 "Example message used as template in MotoHawk models.";

VAL_ 2147490681 Enable 0 "Disabled" 1 "Enabled" ;
```

eerimoq, 2019-04-11:
Bit 31, or MSB, is not part of the frame id in DBC-files. It's 1 for extended frames and 0 for standard frames. At least that's how it is implemented in cantools.

yangliu28, 2019-04-11:
You are absolutely right!
I ran into new problem when I tried to hardcode the unsigned frame ID in header file, and solved it by configuring the frame to be extended. Thanks for your timely help.

closed 2019-04-11 by yangliu28