cantools/cantools #136: C generator support for different bit size of "char"

josko7452, 2019-04-23:
Hi All,

I plan to follow this with an appropriate PR. But first let me explain the situation and see wether you would accept it as an issue.

I expect the target use case of the C generator is to generate code for embedded platforms. 
There is a little problem, not all architectures have CHAR_SIZE == 8 and it can be for example 16, i.e. uint8_t i.e. does not exist for these platforms at all.

For example it is the case of Texas Instrument C2000 DSPs which are actually used quite often in automotive environment.

Currently I am using my own developed coder/decoder (+sequencer) for CAN messages and I'd like to contribute to your generator instead. But I would need to implement this arguably oddness.

So please let me know if this would be accepted as an issue, or you would consider it as unnecessary complication of the codebase...

Thank you
Cheers
Bruno

eerimoq, 2019-04-23:
Hello!

By replacing

``` c
int motohawk_example_message_pack(...)
{
    ...
    if (size < 8u) {
        return (-EINVAL);
    }

    memset(&dst_p[0], 0, 8);
    ...
    return (8);
}
```

with

``` c
int motohawk_example_message_pack(...)
{
    ...
    if (size < 8u * sizeof(uint8_t)) {
        return (-EINVAL);
    }

    memset(&dst_p[0], 0, 8 * sizeof(uint8_t));
    ...
    return (8 * sizeof(uint8_t));
}
```

the generated code would work even if char is 16 bits, right? Just have to extract the lower 8 bits form each element in the packed message before sending it on the bus.

josko7452, 2019-04-23:
Indeed it seems so as you are using explicit masking everywhere..

One thing is that `uint_8t` does not exist there as a type, but that can be work around by typedefing `uint16_t` as `uint8_t` on my side (only danger is that any generator code developed later should not rely on the fact that `uint8_t` is 8bit long, therefore I would propose changing the `uint8_t` to `unsigned char` to be explicit about this fact...)

The `sizeof(uint8_t)` on C2000 (that is if I typedef `uint8_t to uint16_t` will give you 1 though, because the `sizeof()` counts in how many `char`s are in given type...
So instead I'd use `(CHAR_BIT / 8)`

In fact I think there is no need to modify the `memset` bit at all!.
As it will be `size` 8 as it counts chars by 16 bit.
I'll test it and report if it works as I expect (maybe I have overlooked something).

eerimoq, 2019-04-23:
Ok!

eerimoq, 2019-04-24:
Any progress? =)

josko7452, 2019-04-24:
I think you can close it for now and if I encounter specific problem I'll open the specific issue/PR.
It requires some more in-depth testing with the C2000 which I am going to do later...

closed 2019-04-24 by eerimoq