cantools/cantools #138: decode error

mmar2228, 2019-04-29:
Dear all,

I am trying to use a sym file to encode a CAN bus message and decode it. I load successfully the sym file and I encode also the message without problems. However, decoding fails. The following python lines are used to encode and decode the message by using the same database db:

                d_enc = db.encode_message(db.messages[dbcindex].frame_id, d)

                m_dec = db.decode_message(db.messages[dbcindex].frame_id, d_enc)

I get the following error:

m_dec=db.decode_message(db.messages[dbcindex].frame_id, d_enc)
  File "/usr/local/lib/python3.4/dist-packages/cantools/database/can/database.py", line 381, in decode_message
    return message.decode(data, decode_choices, scaling)
  File "/usr/local/lib/python3.4/dist-packages/cantools/database/can/message.py", line 759, in decode
    return self._decode(self._codecs, data, decode_choices, scaling)
  File "/usr/local/lib/python3.4/dist-packages/cantools/database/can/message.py", line 722, in _decode
    scaling)
  File "/usr/local/lib/python3.4/dist-packages/cantools/database/utils.py", line 91, in decode_data
    unpacked.update(formats.little_endian.unpack(data[::-1]))
  File "/usr/local/lib/python3.4/dist-packages/bitstruct.py", line 449, in unpack
    return self.unpack_from(data)
  File "/usr/local/lib/python3.4/dist-packages/bitstruct.py", line 466, in unpack_from
    return {info.name: v for info, v in self.unpack_from_any(data, offset)}
  File "/usr/local/lib/python3.4/dist-packages/bitstruct.py", line 466, in <dictcomp>
    return {info.name: v for info, v in self.unpack_from_any(data, offset)}
  File "/usr/local/lib/python3.4/dist-packages/bitstruct.py", line 332, in unpack_from_any
    yield info, info.unpack(value_bits)
  File "/usr/local/lib/python3.4/dist-packages/bitstruct.py", line 77, in unpack
    return int(bits, 2)
ValueError: invalid literal for int() with base 2: ''

Any idea about this issue??

Thanks you in advance

eerimoq, 2019-04-29:
Please provide a minimal example that reproduces the error.

mmar2228, 2019-04-29:
FormatVersion=6.0 
//Title="test"

{ENUMS}
Enum=DataTypeEnum(16="UCHAR4") // UCHAR4
Enum=E(0="OFF", 1="DIM", 2="on") 
Enum=G(0="Active", 1="standby", 2="paussed",3="finished")

{SIGNALS}
Sig="A" unsigned 8 -h	// Node-ID
Sig="B" DataTypeEnum 8 -h
Sig="C" unsigned 8 -h
Sig=E E 6
Sig=H bit
Sig=I bit
Sig=G G 8
Sig=F unsigned 16 /u:mA
Sig="D" unsigned 8 /ln:"D"

{RECEIVE}

["Status"]
ID=7D1h	
Len=8
Mux="<Empty>" 0,0 0 -t
Sig="A" 0
Sig="B" 8
Sig="C" 16
Sig="D" 24
Sig=E 32
Sig=F 48
Sig=G 40
Sig=H 38
Sig=I 39

mmar2228, 2019-04-29:
That is the sym file. In encode step I create the python dict "d" with the required A-I fields. Enconding is achieved successfully, but not decoding

eerimoq, 2019-04-29:
How did you create the SYM-file? It cannot be loaded with PCAN Symbol Editor 6.

What does it mean to have a multiplexer of zero bits? Is that normal? That likely makes the decoding fail.

eerimoq, 2019-04-30:
I implemented a length check in version 32.18.0. File loading will now fail is any signal has length zero.

mmar2228, 2019-04-30:
Thanks, I was provided that file, so I can not generate another one. I understand it is not correct. I will try to generate another one in dbc format....

mmar2228, 2019-04-30:
I have tried to delete the ""Mux="" 0,0 0 -t" line and I get the following error: error: Invalid syntax at line 26, column 1: ">>!<<Sig="A" 0"

It seems it is neccesary the Mux in order to avoid that fail. How can I manage it?

eerimoq, 2019-04-30:
Download PCAN-Symbol Editor 6 [here](https://www.peak-system.com/Software-APIs.305.0.html?&L=1) and input all information into a new database.

In short, I think you should replace ``"A"`` with ``A``.

mmar2228, 2019-04-30:
It is fine, I will try it. Another related question, I have problems with muxtiplexer...

I use the following dbc file:

VERSION "A"

NS_ :
	NS_DESC_
	CM_
	BA_DEF_
	BA_
	VAL_
	CAT_DEF_
	CAT_
	FILTER
	BA_DEF_DEF_
	EV_DATA_
	ENVVAR_DATA_
	SGTYPE_
	SGTYPE_VAL_
	BA_DEF_SGTYPE_
	BA_SGTYPE_
	SIG_TYPE_REF_
	VAL_TABLE_
	SIG_GROUP_
	SIG_VALTYPE_
	SIGTYPE_VALTYPE_
	BO_TX_BU_
	BA_DEF_REL_
	BA_REL_
	BA_DEF_DEF_REL_
	BU_SG_REL_
	BU_EV_REL_
	BU_BO_REL_
	SG_MUL_VAL_

BS_:

BU_: 

BO_ 800 PM_1_Worklight_Status: 8 Vector__XXX
 SG_ VID_0 M : 16|8@1+ (1,0) [0|0] "" Vector__XXX
 SG_ ID V0 : 0|8@1+ (1,0) [0|0] "" Vector__XXX
 SG_ DT V0 : 8|8@1+ (1,0) [0|0] "" Vector__XXX
 SG_ SC V0 : 16|8@1+ (1,0) [0|0] "" Vector__XXX
 SG_ MC V0 : 24|8@1+ (1,0) [0|0] "" Vector__XXX
 SG_ CT V0 : 32|8@1+ (1,0) [0|0] "" Vector__XXX
 //SG_ VID V0 : 16|8@1+ (1,0) [0|0] "" Vector__XXX
 SG_ WLS V0 : 40|8@1+ (1,0) [0|0] "" Vector__XXX
 SG_ ID V1 : 0|8@1+ (1,0) [0|0] "" Vector__XXX
 SG_ DT V1 : 9|8@1+ (1,0) [0|0] "" Vector__XXX
 SG_ SC V1 : 16|8@1+ (1,0) [0|0] "" Vector__XXX
 SG_ MC V1 : 24|8@1+ (1,0) [0|0] "" Vector__XXX
 SG_ CT V1 : 32|8@1+ (1,0) [0|0] "" Vector__XXX
 //SG_ VID V1 : 16|8@1+ (1,0) [0|0] "" Vector__XXX
 SG_ WLS V1 : 40|8@1+ (1,0) [0|0] "" Vector__XXX
 SG_ ID V2 : 0|8@1+ (1,0) [0|0] "" Vector__XXX
 SG_ DT V2 : 9|8@1+ (1,0) [0|0] "" Vector__XXX
 SG_ SC V2 : 16|8@1+ (1,0) [0|0] "" Vector__XXX
 SG_ MC V2 : 24|8@1+ (1,0) [0|0] "" Vector__XXX
 SG_ CT V2 : 32|8@1+ (1,0) [0|0] "" Vector__XXX
 //SG_ VID V2 : 16|8@1+ (1,0) [0|0] "" Vector__XXX

I get successfully encoding and decoding when I send the message: {'CT': 0, 'DT': 0, 'ID': 0, 'MC': 0, 'SC': 0, 'VID_0': 0, 'WLS': 0} . Note the value 0 for the muxtiplexer signal VID_0.

However, when I encode the following message: {'CT': 0, 'DT': 0, 'ID': 0, 'MC': 0, 'SC': 0, 'VID_0': 1, 'WLS': 0}, (with VID_0 multiplexer signal as 1) the decoding fails:

    m_dec = db.decode_message(db.messages[dbcindex].frame_id, d_enc)
  File "/usr/local/lib/python2.7/dist-packages/cantools/database/can/database.py", line 381, in decode_message
    return message.decode(data, decode_choices, scaling)
  File "/usr/local/lib/python2.7/dist-packages/cantools/database/can/message.py", line 759, in decode
    return self._decode(self._codecs, data, decode_choices, scaling)
  File "/usr/local/lib/python2.7/dist-packages/cantools/database/can/message.py", line 739, in _decode
    scaling))
  File "/usr/local/lib/python2.7/dist-packages/cantools/database/can/message.py", line 722, in _decode
    scaling)
  File "/usr/local/lib/python2.7/dist-packages/cantools/database/utils.py", line 88, in decode_data
    unpacked.update(formats.little_endian.unpack(data[::-1]))
  File "/usr/local/lib/python2.7/dist-packages/bitstruct.py", line 449, in unpack
    return self.unpack_from(data)
  File "/usr/local/lib/python2.7/dist-packages/bitstruct.py", line 466, in unpack_from
    return {info.name: v for info, v in self.unpack_from_any(data, offset)}
  File "/usr/local/lib/python2.7/dist-packages/bitstruct.py", line 466, in <dictcomp>
    return {info.name: v for info, v in self.unpack_from_any(data, offset)}
  File "/usr/local/lib/python2.7/dist-packages/bitstruct.py", line 304, in unpack_from_any
    len(bits)))
bitstruct.Error: unpack requires at least 65 bits to unpack (got 64)

Any idea why multiplexer gives me these problems??

Thanks very much. I am very gratefully

eerimoq, 2019-04-30:
You are compiling with ``strict=False``, right? That means many errors are simply ignored, and it fails later when the database is used.

If you replace ``SG_ VID_0 M : 16|8@1+ (1,0) [0|0] "" Vector__XXX`` with ``SG_ VID_0 M : 48|8@1+ (1,0) [0|0] "" Vector__XXX`` it is probably able to encode and decode the data.

Is this DBC-file actually used in an application? It would be very surprising since the multiplexer signal is overlapping with signals it multiplexes.

mmar2228, 2019-04-30:
Finally, I have removed SC signals since they overlap the multiplexer one. And it works fine. A symbol file similar to this dbc is used in an application, with overlapping signals. I was translating the symbol to dbc because dbc is easier for me and I can detect errors like previous one. 

Thanks very very much!! =)

closed 2019-04-30 by eerimoq