cantools/cantools #143: Install fails on OSX Mojave

PaulWieland, 2019-06-10:
`pip install cantools` results in this error:

```
    ERROR: Complete output from command python setup.py egg_info:
    ERROR: zip_safe flag not set; analyzing archive contents...
    
    Installed /private/var/folders/jl/8cmqmk2s5q39t472_br4xpqm0000gn/T/pip-install-_zU41_/python-can/.eggs/pytest_runner-5.1-py2.7.egg
    /System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/distutils/dist.py:267: UserWarning: Unknown distribution option: 'python_requires'
      warnings.warn(msg)
    error in python-can setup command: 'install_requires' must be a string or list of strings containing valid project/version requirement specifiers; Expected version spec in typing;python_version<"3.5" at ;python_version<"3.5"
    ----------------------------------------
ERROR: Command "python setup.py egg_info" failed with error code 1 in /private/var/folders/jl/8cmqmk2s5q39t472_br4xpqm0000gn/T/pip-install-_zU41_/python-can/

```

It seems like there is a problem with the version of `setuptools` but I cannot figure out how to solve it.

eerimoq, 2019-06-10:
It's the python-can dependency that fails to install. I might make that optional, as it's only used by the monitor subcommand I think.

eerimoq, 2019-06-10:
Hmm, could you open an issue on the [python-can](https://github.com/hardbyte/python-can) project? I would like to keep the dependency to keep the installation simple.

PaulWieland, 2019-06-11:
For anyone else getting stuck with this problem, I worked around this by updating setup tools:

`sudo pip install --upgrade setuptools --user python`

And then installing cantools this way:

`sudo pip install cantools --user python`

cantools is now available:

`WARNING: The script cantools is installed in '/Users/paul/Library/Python/2.7/bin' which is not on PATH.`

closed 2019-06-11 by PaulWieland