cantools/cantools #145: Error while decoding if message is too short

tabidax, 2019-06-18:
I am trying to use the can-tools library to decode the messages in a blf file. However for many messages, I get errors like

“Got only 48 bits, expected 56’,

And indeed the message length according the dbc file is 56, but the message length in the blf file is 48. Canalyzer simply omits the signals corresponding to the missing bits at the end (eg in the above example, the signals corresponding to the last 8 bits in the message are simply not shown), but still decodes the bits at the beginning.

Can the same be achieved using the can-tools library with the decode_message function?

eerimoq, 2019-06-18:
Not right now, no. Why is the message length 48 in the blf file? Isn't that the error?

tabidax, 2019-06-18:
I am not sure, I am using Canalyzer and I would like to switch to can-tools, but I have no influence on the dbc files, which in some cases might contain too many signals. Could you give a hint on where in the can-tools source code I would need to make a custom modification to ignore missing signals?

eerimoq, 2019-06-21:
All signals are decoded at once in cantools. Either all signals in the message will be discarded, or you have to pad the message data to allow decoding of all signals.