cantools/cantools #147: cantools decode not working with multiplexed signals in .sym

thekenu, 2019-06-19:
When I run `cantools decode /path/to/sym`, I get the following error
```
error: The signals alert1 and alert1 are overlapping in message TestAlert.
```
The .`sym` file I am using is as follows:
```
FormatVersion=6.0 // Do not edit this line!
Title="CanMsgs"

{ENUMS}

{SIGNALS}

{SENDRECEIVE}
[TestAlert]
ID=700h
Len=2
CycleTime=1000
Mux=FSM 0,8 1
Var=alert1 unsigned 8,1
Var=alert2 unsigned 9,1
Var=alert3 unsigned 10,1

[TestAlert]
Len=2
CycleTime=1000
Mux=DCM 0,8 2
Var=alert1 bit 8,1
Var=alert2 bit 9,1
Var=alert3 unsigned 10,1
```

The error goes away if I change the `Var` offsets to not overlap between my two multiplexed signals.

eerimoq, 2019-06-21:
Fixed in version 32.19.2, available on PyPI!

closed 2019-06-27 by eerimoq

thekenu, 2019-06-28:
Sorry for the late response! I tested it and the issue went away.