cantools/cantools #150: Problems reading a dbc file, I need help !!!

marcelovieiratecnologia, 2019-08-09:
Error:
Traceback (most recent call last):
  File "C:\PythonProjetos\ArqProducaoAmbulatorial\main.py", line 5, in <module>
    db = cantools.db.load_file(dbfiles[0])
  File "C:\python35\lib\site-packages\cantools\database\__init__.py", line 179, in load_file
    strict)
  File "C:\python35\lib\site-packages\cantools\database\__init__.py", line 247, in load
    strict)
  File "C:\python35\lib\site-packages\cantools\database\__init__.py", line 336, in load_string
    raise UnsupportedDatabaseFormatError(e_arxml, e_dbc, e_kcd, e_sym, e_cdd)
cantools.database.UnsupportedDatabaseFormatError: DBC: "Invalid syntax at line 1, column 1: ">>!<<w~ô% ¡\ufffd       PA_CODUNI  C                   PA_GESTAO  C                   PA_CONDIC  C                   PA_UFMUN   C            PA_REGCT   C                   PA_INCOUT  C                   PA_INCURG  C                   PA_TPUPS   C           PA_TIPPRE  C                   PA_MN_IND  C                   PA_CNPJCPF C                   PA_CNPJMNT C           PA_CNPJ_CC C                   PA_MVM R   C                   PA_CMP EF  C                   PA_PROC_ID C    ""

Process returned 1 (0x1)        execution time : 1.022 s
Pressione qualquer tecla para continuar. . .

eerimoq, 2019-08-09:
Please attach the DBC-file. My guess is that this is an text encoding problem.

marcelovieiratecnologia, 2019-08-12:
Ok, I'm helping myself!
[PAAC1901.zip](https://github.com/eerimoq/cantools/files/3491972/PAAC1901.zip)

eerimoq, 2019-08-12:
That archive does not contain a valid DBC-file.

marcelovieiratecnologia, 2019-08-12:
Ok, thank you!

closed 2019-08-12 by eerimoq