cantools/cantools #152: Issue with parsing ARXML with AUTOSAR_4-2-1.xsd

ForestRupicolous, 2019-08-14:
Hello, I'm trying to parse an ARXML file put only get empty results (no errors).
I tried using the python example to read in a db and also the cantools dump function. I was able to read an .dbc file successfully.
Are ARXML files with the 4.2.1 schema supported and is it possible to enable a verbose output or debugging as I'm not able to share this file (confidential)?
Thanks!

ForestRupicolous, 2019-08-14:
Ok, I just found an ARXML test file with the 4.1.2 schema and it worked fine.

eerimoq, 2019-08-14:
Hello!

I know very little about ARXML and which version(s) that cantools actually supports. Maybe only 4.1.2? =) Support for more versions can be added, but I have no test data so it's hard for me to do correctly.

There is no logging that can be enabled.

ForestRupicolous, 2019-08-20:
Hi, how do you parse the ARXML. Do you use a .xsd as base?
I'm not that familiar with AUTOSAR but found a lot of files on their website. E.g. https://www.autosar.org/fileadmin/user_upload/standards/classic/4-2/AUTOSAR_TR_XMLSchemaSupplement.zip
from https://www.autosar.org/nc/document-search/

chaooe, 2021-06-24:
I also found this issue with no errors. Also without no messages in it.
{'_messages': [], '_nodes': [], '_buses': [], '_name_to_message': {}, '_frame_id_to_message': {}, '_version': None, '_dbc': None, '_frame_id_mask': 4294967295, '_strict': True}

http://autosar.org/schema/r4.0

H4rryK4ne, 2021-06-24:
I have the same issue, but I suppose, that's because in my file there are no CAN signals, just Ethernet (I-SIGNAL) ;)

andlaus, 2021-06-24:
cantools only supports those parts of ARXML that deal with CAN frame descriptions (i.e., the in stuff described by CAN-FRAME-TRIGGERING elements). Support for this should be pretty robust by now, though. Are you sure your file includes descriptions of CAN frames? if yes, I might have a look if you provide me such an ARXML file for testing.

H4rryK4ne, 2021-06-25:
@andlaus as I thought already, there are no CAN-FRAME-TRIGGERING elements in my ARXML

I have another ARXML with CAN-FRAME-TRIGGERING which is throwing an exception during load. But I will open another issue, as soon as I can compose a minimal example ARXML to reproduce the error.

chaooe, 2021-06-25:
I got wordaround to fix this issue. I use canmatrix which have canconvert tools to convert to dbc.

canconvert xxx.arxml 2.dbc

```
cantools.database.load_file("2.dbc")
```
It works.