cantools/cantools #157: use frame().msg from can package to decode dbc using message_decode from cantools

ayejinglebell, 2019-09-05:
I want to use the **can_msgs/Frame.msg** for decoding can messages using **db.decode_message(Frame.id, Frame.data)** but it is giving error.

`def callback(Frame):`
`rospy.loginfo(rospy.get_caller_id() + "I heard %s", Frame.data)`
`Temp = db.decode_message(Frame.id, Frame.data)`
`temp.data= Temp`
`pub.publish(temp)`

I have (specially encoded) canbus data, published already on ros, as a bunch of can_msgs.Frame msgs. You want to receive the ros Frame msg, decode it, and you publish it as rostopics.

Error:
File "safa.py", line 42, in callback temp = db.decode_message(Frame.id, Frame.data) 
File "build/bdist.linux-x86_64/egg/cantools/database/can/database.py", line 379, in decode_message 
message = self._name_to_message[frame_id_or_name] 
KeyError: 10

chrissunny94, 2019-09-12:
`#!/usr/bin/python
 #encoding:utf-8

 import cantools
 import rospy
 from can import Message
 from can_msgs.msg import Frame
 from typing import Optional
 from std_msgs.msg import String

 frame = Frame()

 dbc_file = open("/home/cth4kor/Desktop/DBC/CURTIS_CAN.dbc", "r", encoding="iso-8859-1")
 db = cantools.database.load(dbc_file)
 #db = cantools.database.load_file('/home/cth4kor/Desktop/DBC/CURTIS_CAN.dbc')

 #print(db.get_message_by_name('CURTIS_ControlMessage').signals)
 #print(db.get_message_by_name('CURTIS_InformationMisc').signals)
 print(db.get_message_by_name('CURTIS_Information').signals)

 #message = can_bus.recv()

 #print(message.data, message.arbitration_id)
 print("\n\n")
 print(db.messages)

 pub = rospy.Publisher('CURTIS_VehicleMotorSpeed',String, queue_size=10)

 def callback(Frame):
     rospy.loginfo("\n"+rospy.get_caller_id() + " I heard %s %s", Frame.id,Frame.data)
     output = db.decode_message(Frame.id, Frame.data)
     CURTIS_VehicleMotorSpeed = output.get('CURTIS_VehicleMotorSpeed')
     print(CURTIS_VehicleMotorSpeed)
     
 def listener():
    rospy.init_node('CAN', anonymous= True)
    rospy.Subscriber("/CAN_frame/input",String,callback)

 if __name__ == '__main__':
     rospy.init_node('CAN',anonymous=True)
     rospy.Subscriber("/CAN_frame/input", Frame, callback)
     rospy.spin()
`

Stalla-SR, 2020-08-14:
Hello chrissunny94,
Thank you for the .py script you provided. 

I am currently playing around with translating CAN messages from ROS trough cantools and publishing back to ROS and I tried the script, however, I'm having trouble with the message decoding. 

From the line:  dbc_file = open("/Path/to/my/DBC/file.dbc", "r", encoding="iso-8859-1") i get 
/CAN_4815_1597393294949 I heard 280 b' \x00\x00\x00\x00\x00\x00\x00\x00' 
None

And when I do a rostopic echo /CURTIS_VehicleMotorSpeed I get:
---
data: "None"
---

This specific message in the ,dbc file is:
BO_ 280 Wheel_Data_3: 8 Vector__XXX
 SG_ RR_WheelSpeed_Rpm : 55|16@0- (0.1,0) [-3276.8|3276.7] "rpm" Vector__XXX
 SG_ RL_WheelSpeed_Rpm : 39|16@0- (0.1,0) [-3276.8|3276.7] "rpm" Vector__XXX
 SG_ FR_WheelSpeed_Rpm : 23|16@0- (0.1,0) [-3276.8|3276.7] "rpm" Vector__XXX
 SG_ FL_WheelSpeed_Rpm : 7|16@0- (0.1,0) [-3276.8|3276.7] "rpm" Vector__XXX

I've tried change the encoding to CP1252 for the Vector dbc file but I get the same result...

Stalla-SR, 2020-08-14:
I got it working for me in ROS with some little modifications to the above code:

#!/usr/bin/python3
#encoding:utf-8

import cantools

import rospy
from can import Message
from can_msgs.msg import Frame
from typing import Optional
from std_msgs.msg import String

frame = Frame()

dbc_file = open("/Path/to/my/DBC/file.dbc", "r", encoding="iso-8859-1")
db = cantools.database.load(dbc_file)
pub = rospy.Publisher('translated_CAN_messages',String, queue_size=10)

def callback(Frame):
    rospy.loginfo("\n"+rospy.get_caller_id() + " I heard %s %s", Frame.id,Frame.data)
    output = db.decode_message(Frame.id, Frame.data)
    translated_CAN_messages = (Frame.id, Frame.data, output)
    pub.publish(str(translated_CAN_messages))

def listener():
    rospy.init_node('CAN', anonymous= True)
    rospy.Subscriber("CAN_messages_from_ROS",String,callback)

if __name__ == '__main__':
    rospy.init_node('CAN',anonymous=True)
    rospy.Subscriber("CAN_messages_from_ROS", Frame, callback)
    rospy.spin()

Thank you chrissunny94 for the python script and eerimoq for creating the package.

pinnakarahul, 2021-09-23:
hello

can any one say about me canopen..?

how can i go through that...?

AlexDLSy, 2021-09-23:
@pinnakarahul Please do not ask unrelated questions in separate topics from your issue.
I will give you some insight in #344.