cantools/cantools #158: The code given in documentation as example fails to decode

chrissunny94, 2019-09-09:
I have just tried to use the given code in the documentation to see how the decoding works but I get an error every time.

Nothing actually gets decoded. I am generating data in vcan0 using the cansend command.

`#!/usr/bin/python`
`#encoding:utf-8`

`import can`
`import cantools`

`db=cantools.database.load_file('/home/cth4kor/cantools/tests/files/dbc/motohawk.dbc')`
`can_bus=can.interface.Bus('vcan0', bustype='socketcan')`
`example_message =db.get_message_by_name('ExampleMessage')`
`data= example_message.encode({'Temperature': 250.1, 'AverageRadius': 3.2, 'Enable': 1})`
`message= can.Message(arbitration_id=example_message.frame_id, data=data)`
`can_bus.send(message)`
`message =can_bus.recv()`
`print(db.decode_message(message.arbitration_id,message.data))`

Error I am getting:

Traceback (most recent call last):
  File "/usr/local/lib/python3.5/dist-packages/cantools-32.20.0-py3.5.egg/cantools/database/can/database.py", line 377, in decode_message
KeyError: 1154

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "new.py", line 19, in <module>
    print(db.decode_message(message.arbitration_id, message.data))
  File "/usr/local/lib/python3.5/dist-packages/cantools-32.20.0-py3.5.egg/cantools/database/can/database.py", line 379, in decode_message
KeyError: 1154

eerimoq, 2019-09-09:
Looks like you are sending a CAN frame with the wrong id (1154, or 0x482).

Try `cansend vcan0 1f0#0000000000000000`.

chrissunny94, 2019-09-09:
Hi is it possible to decode without encoding? I am using can_msgs.Frame.msg to receive the already available signals using ROS. I just want to decode the signals using the DBC provided to get it in a human readable form.

`#!/usr/bin/python`
`#encoding:utf-8`

`import cantools`

`import rospy`
`from can import Message`
`from can_msgs.msg import Frame`
`from typing import Optional`
`from std_msgs.msg import String`

`frame = Frame()`
`db= cantools.database.load_file('/home/cth4kor/Desktop/DBC/CURTIS_CAN.dbc')`

`pub= rospy.Publisher('vehicle_temp', Frame, queue_size=10)`

`def callback(Frame):`
    `rospy.loginfo(rospy.get_caller_id()+ "I heard %s", Frame.data)`
     `temp= db.decode_message(Frame.id, `Frame.data)`
    `pub.publish(temp)`

`def listener():`
   `rospy.init_node('CAN', anonymous= True)`
    `rospy.Subscriber("/CAN_frame/input",String,callback)`

`if__name__ == '__main__':`
    `rospy.Subscriber("listener",Frame, callback)`
    `#spin() simply keeps python from exiting until this node is stopped`
    `try:`
        `listener()`
         `except rospy.ROSInterruptException:`
          `pass`
          `rospy.spin()`

**Error:**
[ERROR] [1568012732.766221]: bad callback: <function callback at 0x7f05ad9d4e18>
Traceback (most recent call last):
  File "/usr/local/lib/python3.5/dist-packages/cantools-32.20.0-py3.5.egg/cantools/database/can/database.py", line 377, in decode_message
    message = self._frame_id_to_message[frame_id_or_name]
KeyError: 0

eerimoq, 2019-09-09:
Sorry, I don't hav etime to help further.

Uch21, 2019-09-11:
Hi chrissunny94,
how does your data look like? Is it a bytearray? Something like: bytearray(b'\x00\x01\x02\x03\x04\x05\x06\x07').

And are you sure that your message id is defined in the dbc file?

closed 2019-09-12 by chrissunny94