cantools/cantools #161: DBC Min 0 and Max 0 Results in None Being Set

aquatsr, 2019-10-07:
Hello,

For DBC entries with signals having min and max both equal to 0, the signal object gets set with `_minimum = None` and `_maximum = None`.

Is there a way to retain what is present in the DBC entry upon load? Possible to document this behavior if it's considered normal/desirable?

Thanks

PUNITPNT, 2021-08-25:
hi @aquatsr 

I also have same problem.

U got any solution to this ?
how you resolved this issue ?

tanasovich, 2023-03-21:
I also have the same problem.

aquatsr, 2023-03-22:
Hey @PUNITPNT @tanasovich 

Assuming you know what the min/max for each signal having `None`/`None` is, you'll need to loop through the signals for each message in the cantools database and update the min/max for each. If your min/max is 0/0, you're essentially undo-ing what cantools does when loading the CAN database file.

Hope that helps.