cantools/cantools #162: Merging DBCs with different sets of attributes creates invalid output

geo4711, 2019-10-30:
I tried to merge two dbcs, just like this:
`dbc = cantools.database.load_file("basefile.dbc")`
`dbc.add_dbc_file("another.dbc")`
`cantools.database.dump_file(dbc, "merged.dbc")`
 

Unfortunately, the created file cannot be read (by Vectors CANdb++.exe, which reports the use of an unknown message attribute). Debugging the file in text mode shows, that the merged files contails a line like:
`BA_ "GenMsgILSupport" BO_ <myID> 0;`
But there should be another two lines to define the attribute itself and its default value, like that:
`BA_DEF_ BO_  "GenMsgILSupport" ...;`
`BA_DEF_DEF_  "GenMsgILSupport" ...;

In my case, "basefile.dbc" does not contain any custom attributes at all, "another.dbc" does.
"basefile.dbc" had been created with cantools from scratch`.

geo4711, 2019-10-30:
The function which adds the dbc to the current database is `add_dbc_string()` in `database.py`.

The current code seems to add the message, **replace** some other stuff, but makes no changes for attributes:
```
    def add_dbc_string(self, string):
        database = dbc.load_string(string, self._strict)

        self._messages += database.messages
        self._nodes = database.nodes
        self._buses = database.buses
        self._version = database.version
        self._dbc = database.dbc
        self.refresh()
```

Proposed solution:
- add new dbc's lines `BA_DEF_`and `BA_DEF_DEF_` as well
- check for duplicates
-- same BA_DEF_/BA_DEF_DEF_ with identical contents are fine
-- same BA_DEF_/BA_DEF_DEF_ with different contents should lead to complete abort of the merge process
- nodes should be merged in a similar way (check for duplicates by name)

Maybe same logic for other dbc sections? I'm not sure what BA_DEF_REL_/BA_DEF_DEF_REL_ is used for ...

MerrimanInd, 2020-09-21:
I've run into this same issue, trying to merge DBCs. I agree with your recommendation for behavior: an inclusive OR for buses, attributes, nodes, etc. Perhaps it would be more beneficial to add dedicated merge functions to preserve backwards compatibility and for clarity? I'd like to start working on a function for merging two database objects maybe in addition to a merge function similar to the existing add_dbc_string() function that pulls directly from a file.

geo4711, 2020-09-22:
I had started with that last year, but cancelled it. The main item was to deal with conflicts. E. g. same attributes in both files, but different default values. Maybe even different types (int vs. string etc.). For a proper solution reflecting all that, it will not be done in a few hours.
I had thought about an interactive mode (where the user can decide if file 1 or file 2 shall "win", but I'm not sure if user interaction is compatible with the project's concept. And though, it would still be too much work for me.

My use case when creating this issue was just to import one message from dbc_2 into dbc_1. And in that concrete case, it was much more easy to just skip all the attributes from that message, as it was only additional information not necessary in dbc_1.

But back to your proposal:
I agree: separate (new) functions should be better, to be backwards compatible.
Maybe split between functions for "checking" and "doing":
* checking: test, if attributes of both databases have conflicts (names, type, value ranges, default values, ...); maybe list all conflicts
* doing: merge the files, with some parameters that control conflict handling

MerrimanInd, 2020-09-22:
Yeah I agree on all counts, both the issues with this and your approach. I'll work on two functions: a merge function with an option for 'stop on conflicts' vs 'overwrite conflicts' and a check function that outputs a database object containing only the conflicted elements so they can be iterated through and remedied programmatically.

This isn't my main job function so don't expect fast progress but I'm committed!

ptoews, 2021-08-10:
I'm having a similar issue while merging DBC files, where the resulting DBC cannot even be loaded again by cantools because of a KeyError. @MerrimanInd your plan sounds promising, anything new about it?

closed 2026-10-29 by andlaus