cantools/cantools #167: Long object names in DBC not fully supported

geo4711, 2019-10-30:
I try to rename certain objects in a dbc file (messages and signals).
The dbc supports a "normal" symbol length up to 32 chars. If longer names shall be used, that seems to be done "in the background" by special attributes, e. g. with a line like that:
`BA_ "SystemSignalLongSymbol" SG_ <frame_id> <signal name up to 32 chars> " <long signal name>";`
Reading that attributes works fine (always reports the correct name, i. e. the long name, if applicable).
Writing/Changing that attribute however fails, or let's say, it seems to work, but unfortunately "forgets" to change the special attribute.
When opening the modified dbc again, the original name is still there ...

eerimoq, 2019-10-30:
I'm sorry, but I'll not have time looking into the issues you created. Any help fixing the bugs is appreciated.

geo4711, 2019-10-30:
Thanks for your quick reply :-)
And sorry for my very short issue report. I'm not very familiar with the module's code, but I just spent some hours to understand the main logic. I hope I've found the corresponding code snippet now.

It is function `_dump_attributes(database)` in `dbc.py`.
The function iterates following object types, where long symbolic names could appear:
- nodes
- messages
- signals (for each message)

The following checks/measures should be added for each of the loops:
- Decide, if the objects name can be encoded "directly" or not, i. e. the attribute `SystemSignalLongSymbol` will be neccessary. One reason for that is `len(<object's name>) > 32`, but maybe there are other conditions which require to use the attribute (special characters or just a blank in the object's name?).
- If att is neccessary: update its value, or add it, if not existing yet.
- Else: Remove the attribute for that object, if existing (so that it wont be written to the file anymore).

The following code (for object type `signal`) could do that (however it's moreless pseudo code, as I'm not sure how to access the data structure in a proper way ...):
```
DBC_MAX_SYMBOL_LENGTH_STANDARD = 32

if len(signal.name) > DBC_MAX_SYMBOL_LENGTH_STANDARD or <other reasons like special char>:
    # --> ensure correct attribute value to store the complete name
    if "SystemSignalLongSymbol" in [att.kind for att in signal.dbc.attributes]:
        # attribute "long name" already existing --> update value
        signal.dbc.attributes['SystemSignalLongSymbol'] = signal.name
    else:
        # attribute value not existing --> add it:
        signal.dbc.attributes.append[kind='SystemSignalLongSymbol', value=signal.name, ...]
else:
    # --> delete this attribute, if existing
    try:
        signals.dbc.attributes.remove[kind='SystemSignalLongSymbol']
    except:
        pass
```

I hope having found the right position and code and a good suggestion for a solution.
I'll proceed similarly for my other today's tickets ...

geo4711, 2019-11-10:
I've looked in the code more deeply now.
It seems that _reading_ long names from dbc is supported quite well, but _writing_ to dbc is limited.
Before making proposals for code adaptions, I'd like to get your opition about the best way to proceed.

The module is capable of handling different types of databas files. The only one I've been using is dbc. So I'm not familiar with specifications of arxml, sym, ... and their syntax restrictions.
However, the internal data structure of _cantools_ seems to be adopted from dbc (dbc.dbcSpecifics, name mappings for names > 32chars,  ...).

After being loaded, all objects can be accessed/found by their name (without having to know if is a short or a long name). Still, their attributes for long names are kept in the attributes lists.

My favourite attempt would be to skip (or at least ignore) those "long name attributes", and recreate them only when generating a dbc string again. That would be a proper way to create a long name attribute exactly for that objects where it is neccessary.
And if other file formats have other restrictions, those converter function could have different suitable methods for that.

Another aspect are different kinds of references between object types (e. g. sender node of a message). For an automatic update ("refactoring") I assume one will need a method from _database context,_ e. g.:

```
def dbc.rename_node(self, old_node_name, new_node_name):
    node = self.get_node_by_name(old_node_name)
    node.name = new_node_name
    for msg in self._messages:
        if old_node_name in msg.senders:
            #replace sender ...
````                

instead of just writing `node.name = new_node_name`.

From node context it is not possible to get a list of all related messages, is it? I think the node object does not now about the database object it is contained in ...
(I've read in some python forums that there are ways to get the parent object in a cascaded data structure. But neither am I used to that technique nor could I find it already in the code. However, using this concept could make the rest of the code easier --> renaming objects could be done from 'everywhere', and the object itself could care about the rest, i. e. when changing an object's name, its method decorated with `@name.setter` could do the rest ...)

I would be interested in your opition about this topic :-)

eerimoq, 2019-11-10:
I'm sorry, but I don't have time to take part in the discussion. I trust you in your observations and will merge any PR that improves cantools.

geo4711, 2019-11-12:
Thanks for your confidence, I'll do my best and come up with a PR.

PR #179 (2020-01-05): Main change:
Closes #167: DBC export: improved handling for long symbol names (nodes, messages, signals); prepared for env_vars, too (but currently they are not exported to dbc at all);

Minor improvements:
- dbc version string (add write access, map default value None to '')
- correct BA Enum for kind = None (as already implemented for other kinds)
- improve robustness of 1 testcase dealing with binary dbc comparision: tolerate differences caused by os specific newlines
- dbc import: convert long names from environment variables, too.
- dbc import/export: replace "magic strings" by string con [... cut]
commit f2e4de20 (2020-01-05): Main change:
Closes #167: DBC export: improved handling for long symbol names (nodes, messages, signals); prepared for env_vars, too (but currently they are not exported to dbc at all);

Minor improvements:
- dbc version string (add write access, map default value None to '')
- correct BA Enum for kind = None (as already implemented for other kinds)
- improve robustness of 1 testcase dealing with binary dbc comparision: tolerate differences caused by os specific newlines
- dbc import: convert long names from environment variables, too.
- dbc import/export: replace "magic strings" by string con [... cut]
closed 2020-01-05 by eerimoq