cantools/cantools #172: cantools don't work in python3

loci41, 2019-11-13:
Hello, i wrote a python-programm which can read and save CAN-data on a raspberry pi 3B+. When I start it in the terminal of the raspberry in python, it works. But when i start it in python3, it don't work. 

ERROR:
File "program.py", line11, in <module>
      import cantools
ModuleNotFoundError: No module named 'cantools'

eerimoq, 2019-11-13:
Looks like you havn't installed cantools. I've not used Raspberry PI myself, but I think you can install it with pip, or possibly pip3?

loci41, 2019-11-13:
thanks for your fast advice. I found my mistake. 
I just installed cantools for python --> pip install cantools
now i installed it for python3 --> pip3 install python

closed 2019-11-13 by loci41