cantools/cantools #173: Reading of CAN Signal attributes not supported

mahmad123, 2019-12-05:
### Background
DBC files support the creation of custom attributes that can be associated with Messages, Signals or Nodes.  These are useful when defining custom properties not supported by default signal parameters. 

### Description of Issue
Current version of this tool does not support reading of attributes.   

### Proposed Change
Read in all attributes for messages, signals and nodes when parsing dbc file.

geo4711, 2019-12-07:
This function is already available.
There is a generic container for dbc specific attributes, applied to nodes, messages and signals.
You can access it with any kind of these objects like in the follwing example:

```
db = cantools.database.load_file(<your_dbc>)
msg = db.get_message_by_frame_id(<your_id>)
msg.dbc.attributes
sig = msg.signals[<your signal>]
sig.dbc.attributes
```

You will receive an ordered dictionary.
Try one of the test dbcs delivered with the package, e. g. "attributes.dbc", and you should get the following results:
```
In [1]: db = cantools.database.load_file(r"tests\files\dbc\attributes.dbc")
In [2]: db.get_message_by_frame_id(42).signals[0].dbc.attributes
Out[2]: OrderedDict([('GenSigSendType', attribute('GenSigSendType', 1))])

In [3]: db.get_message_by_frame_id(42).signals[0].dbc.attributes['GenSigSendType']
Out[3]: attribute('GenSigSendType', 1)

In [4]: db.get_message_by_frame_id(42).signals[0].dbc.attributes['GenSigSendType'].value
Out[4]: 1
```

I hope this solves your problem. If yes, please close your issue.

One additional hint:
If you want to **add** your own attributes (i. e. attributes that have not been defined in the file you've read), be sure to declare the attribute and its default value on database level (otherwise, the attribute value for your object can't be referenced, and dumping the modified database will fail.)

mahmad123, 2020-01-30:
Sorry for taking so long to get back to you.  I have tried this out and you are indeed correct (not that you needed me to tell you). I was confused on how to access this data.  Thanks for the guidance.

closed 2020-01-30 by mahmad123

mahmad123, 2020-01-30:
I actually had another question on this topic.  I am using this module in a MATLAB script and found that if i access the *CAN signal* attribute through another method, i will have this issue.  Using the method you had (by `using get_message_by_name` method) works for me.  But if i try to index through message, then down to signal, i don't find any attributes (blank).  No errors, just empty attributes field.

Below is the method i use that is successful and the one that fails.  Could you have a look an let me know why i one does work but the other does. 

#### Failed Method
```matlab
% MATLAB Script
py.importlib.import_module('cantools');
db = py.cantools.database.Database();
db.add_dbc_file('example.dbc');

s = {};
dbcMsgs = cell(db.messages);
 for i=1:length(dbcMsgs)
     sigs = cell(dbcMsgs{i}.signals);
     for j=1:length(sigs)
        sigs{j}.dbc.attributes;
     end
end
```

### Successful Method
I do have success with this method though.  

```matlab
py.importlib.import_module('cantools');
db = py.cantools.database.Database();
db.add_dbc_file('example.dbc');

msg = db.get_message_by_name('example_msg');
sig = msg.get_signal_by_name('example_sig);
att = struct(sig.dbc.attributes);
string(att.EnumExample.value)
```

kailasgogate, 2023-08-04:
Hello, When I use msg.dbc.attributes, it gives some of the attributes defined for that particular message.
When I checked further, it looks like it returns dict of attributes which has value other than default value.

Is it possible to read all the attributes? regardless of if it has default value or not?

martinmain, 2023-10-14:
Hi I am trying to extract an enum formula from a user attribute added to my DBC file but struggling to see how I can access the result I require, (which is the formula under the enum  item 3), best I can achieve is:

attribute('RealDash', 3) for my signal 36. using cansignal.dbc.attributes.get('RealDash')

from DBC file:

BA_DEF_ SG_  "RealDash" ENUM  "","V>>4","V>>2","V &amp;127","V &amp: 128";
BA_DEF_DEF_  "RealDash" "";
BA_ "RealDash" SG_ 36 Yaw_Rate 3;

Any advice most welcome

juleq, 2023-10-14:
What was the output and what was expected?

martinmain, 2023-10-14:
@juleq 

The output was attribute('RealDash', 3)

I wanted "V &127" from enum list

juleq, 2023-10-14:
Ah, ok. So you want to resolve the enum. Off the top of my head I could not tell if there is API for that. But the enum should be in the dbc specifics so you could resolve it yourself.

martinmain, 2023-10-14:
Thanks, I figured this might be the case, instead I have changed the enum values to become meaningful text for the dropdown menu in the Kvaser Editor and implemented the values I wanted in python using case and then using the following statement to correctly assigne the case selection

 att_index = str(cansignal.dbc.attributes.get('RealDash'))[22:23]

l-schilling, 2023-11-15:
> This function is already available. There is a generic container for dbc specific attributes, applied to nodes, messages and signals. You can access it with any kind of these objects like in the [...]

This is very helpful. While it is inside the documentation of "Functions and classe" maybe including an example would be helpful?