cantools/cantools #181: Can't find documentation on Message object layout

ttepatti, 2020-01-21:
I imported a database using the same code that was utilized in the example documentation:
`db.cantools.database.load_file('motohawk.dbc')
db.messages`

This returns the following list:
`[message('ExampleMessage', 0x1f0, False, 8, 'Example message used as a template in MotoHawk models.')]`

This is great, but I can't find any documentation that explains what order the Message object's values are in. Obviously I can discern that the first is the message name, and second is the Frame ID, but I have no idea what the third field ("false") relates to.

If someone could point me to documentation on this object type or the source code where the string is constructed before being returned, that would be great.

Thank you!

eerimoq, 2020-01-21:
https://cantools.readthedocs.io/en/latest/#cantools.database.can.Message

ttepatti, 2020-01-21:
> https://cantools.readthedocs.io/en/latest/#cantools.database.can.Message

Omg, embarrassing. I swear I Ctrl+F'd everything in that document, but I guess not. Thank you!

closed 2020-01-21 by ttepatti