cantools/cantools #187: DBC Attribute Types

DaGuich, 2020-02-28:
In a DBC file attributes are bound to object types like Signals, Nodes, Networks and so on. Currently this implementation does not reflect this binding.

bandeezy, 2020-07-23:
Where do you see this? I'm by no means an expert regarding `cantools`, but it seems that attributes are assigned to their signals, messages, and nodes. See [dbc.py#L1185](https://github.com/eerimoq/cantools/blob/master/cantools/database/can/formats/dbc.py#L1185) as an example.

DaGuich, 2020-09-03:
At least in the Vector tools (CANdb++) you are declaring attributes (with a default value) for a object type, which are then the global default. These attributes are then open to refine for the object (message, signal, ...) in question.

bandeezy, 2020-09-05:
The current implementation does bind attributes to their corresponding database, nodes, messages, and signals. Have you checked recently? I'm currently using version `32.20.0` and I can access the attributes in question just fine.

It's also generally helpful to provide some more context when creating an issue. Are you trying to parse a DBC with cantools? When a database object is instantiated, node, message, and signal attributes can be accessed via the `dbc` object contained within them. See a brief example I quickly threw together below, just replace `DBC_FILE` and `NODE_STRING` with the path to your DBC file and the node of interest, respectively:
```
from cantools import db

DBC_FILE = "example.dbc"
NODE_STRING = "example_node"

database = db.load_file(DBC_FILE)

# database attributes
if not database.dbc.attributes:
    print(f"No database attributes")
else:
    for _, db_attribute in database.dbc.attributes.items():
        print(f"Database attribute: {db_attribute}")

node = database.get_node_by_name(NODE_STRING)
if not node.dbc.attributes:
    print(f"No node attributes")
else:
    for _, node_attribute in node.dbc.attributes.items():
        print(f"Node attribute: {node_attribute}")

for msg in database.messages:
    # message attributes
    if not msg.dbc.attributes:
        print(f"No message attributes for \"{msg.name}\"")
    else:
        for _, msg_attribute in msg.dbc.attributes.items():
            print(f"Message attribute for frame \"{msg.name}\": {msg_attribute}")
            # signal attributes
            for sig in msg.signals:
                if not sig.dbc.attributes:
                    print(f"No signal attributes for \"{sig.name}\"")
                else:
                    for _, sig_attribute in sig.dbc.attributes.items():
                        print(f"Signal attribute for signal \"{sig.name}\": {sig_attribute}")
```