cantools/cantools #188: Only decode messages in DBC

StrangeCalibur, 2020-03-20:
First of all, this is an amazing tool and thanks for all the hard work that's been put into it.

Second of all, I am sorry that I can't share files for this but unfortuantly they are all NDA.

Is it possible to only decode messages that have been brought in with the dbc and ignore everything else? 
At the moment I am trying to filter them before I get to the decode stage but I need to manually wait for each error, then copy the id to the blacklist, sadly probably hundreds of ID floating about like this which would take forever to filter out. 

> #Log file
> filename = "sorry this file is NDA.blf"
> 
> #DBC Files
> dbfiles = [
>     "sorry this file is NDA.dbc",
>     "sorry this file is NDA.dbc"
> ]
> 
> #Load first DBC file
> db = cantools.database.load_file(dbfiles[0], 'dbc', 'cp1252', None, False)
> 
> #Append rest of DBC files to dbc component
> for i in range(1, len(dbfiles)):
>     db.add_dbc_file(dbfiles[i])
> 
> 
> 
> #db.messages
> #print(db.messages)
> 
> #Convert file to list
> log = list(can.BLFReader(filename))
> 
> not_in_DBC = [419372071,
>               419215091,
>               419321903,
>               419321903,
>               217055747,
>               5,
>               419320563,
>               419335667,
>               419362803,
>               419367411,
>               418389476,
>               419321920,
>               419359279,
>               418384696,
>               419320051,
>               419392840,
>               419409480,
>               419427143,
>               419375943,
>               419392583,
>               215154471,
>               419409223,
>               418328359,
>               419361024,
>               419361063,
>               419376679,
>               418326319,
>               419387687,
>               419387943,
>               419426855,
>               217056000,
>               419361280,
>               419348235,
>               418389259,
>               418382091,
>               419319563,
>               419351112,
>               419376200,
>               419376968,
>               419391816,
>               419372327
> 
>               ]
> 
> 
> 
> 
> 
> for msg in log:
>     if msg.arbitration_id in not_in_DBC:
>         print(f"Not in DBC file: {msg}")
>     else:
>         message = db.decode_message(msg.arbitration_id, msg.data, False, True)
>         print(message)
> 

Output:
{this message is valid but NDA so replacing with this text}
Not in DBC file: Timestamp: 1580302646.690811    ID: 18ff1d27    X                DLC:  8    10 00 50 03 00 00 00 06     Channel: 0
Not in DBC file: Timestamp: 1580302646.691096    ID: 18ff5927    X                DLC:  8    ca ca fa fa 00 00 00 65     Channel: 0
Not in DBC file: Timestamp: 1580302646.691390    ID: 18ff5a27    X                DLC:  8    00 00 00 00 00 00 00 65     Channel: 0
Traceback (most recent call last):
  File "C:\Users\a1069763\PycharmProjects\RF_Harvest_Verification\venv\lib\site-packages\cantools\database\can\database.py", line 385, in decode_message
    message = self._frame_id_to_message[frame_id_or_name]
KeyError: 486475047

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:/Users/a1069763/PycharmProjects/NDA_Verification/Sketch Sheet.py", line 85, in <module>
    message = db.decode_message(msg.arbitration_id, msg.data, False, True)
  File "C:\Users\a1069763\PycharmProjects\NDA_Verification\venv\lib\site-packages\cantools\database\can\database.py", line 387, in decode_message
    message = self._name_to_message[frame_id_or_name]
KeyError: 486475047

Process finished with exit code 1

eerimoq, 2020-03-21:
Try

``` python
for msg in log:
    try:
        print(db.decode_message(msg.arbitration_id, msg.data, False, True))
    except KeyError:
        pass
```

StrangeCalibur, 2020-03-21:
Worked a charm! My hero! Obviously I am not the best dev in the world, but as someone working in the industry, is there anything I can do to support your development? I work with vector hardware and software on a daily basis so if there is any testing you need or verification of various features I would love to lend a hand.

closed 2020-03-21 by StrangeCalibur

eerimoq, 2020-03-21:
Great! =)

Anything that improves the library is welcome. I can't think of any particular test that is needed right now, sorry.