cantools/cantools #189: KeyError during loading of .cdd file

ZaferBuker, 2020-03-21:
**Source code:**

import cantools

print(cantools.database.load_file('x.cdd')) 

**Error:**

Traceback (most recent call last):
  File "C:/Users/ZX/.PyCharmCE2018.3/config/scratches/scratch.py", line 3, in <module>
    print(cantools.database.load_file('C:/Users/ZX/Desktop/CAN Simulation/diag/denbe.cdd'))
  File "C:\Users\ZX\AppData\Local\Programs\Python\Python37-32\lib\site-packages\cantools\database\__init__.py", line 180, in load_file
    strict)
  File "C:\Users\ZX\AppData\Local\Programs\Python\Python37-32\lib\site-packages\cantools\database\__init__.py", line 255, in load
    strict)
  File "C:\Users\ZX\AppData\Local\Programs\Python\Python37-32\lib\site-packages\cantools\database\__init__.py", line 339, in load_string
    db.add_cdd_string(string)
  File "C:\Users\ZX\AppData\Local\Programs\Python\Python37-32\lib\site-packages\cantools\database\diagnostics\database.py", line 60, in add_cdd_string
    database = cdd.load_string(string)
  File "C:\Users\ZX\AppData\Local\Programs\Python\Python37-32\lib\site-packages\cantools\database\diagnostics\formats\cdd.py", line 196, in load_string
    data_types)
  File "C:\Users\ZX\AppData\Local\Programs\Python\Python37-32\lib\site-packages\cantools\database\diagnostics\formats\cdd.py", line 166, in _load_did_element
    data_types)
  File "C:\Users\ZX\AppData\Local\Programs\Python\Python37-32\lib\site-packages\cantools\database\diagnostics\formats\cdd.py", line 139, in _load_data_element
    data_type = data_types[data.attrib['dtref']]
KeyError: '_0B2052D0'

Process finished with exit code 1

eerimoq, 2020-03-25:
Could you by any chance attach the cdd-file? It's very hard to fix the problem without it.

ZaferBuker, 2020-03-26:
Sorry but I couldn't find any template file.

ZaferBuker, 2020-03-26:
Sorry because I couldn't find any template file.

closed 2020-05-11 by ZaferBuker

govind-a-m, 2023-10-31:
add below lines in _load_data_types(cdd.py) method
    types += ecu_doc.findall('DATATYPES/NUMITERDT')
    types += ecu_doc.findall('DATATYPES/MUXDT')

Carlopett, 2024-01-15:
I faced the same issue. The solution proposed by @govind-a-m does not solve the issue in my case. Any idea?

govind-a-m, 2024-01-15:
> I faced the same issue. The solution proposed by @govind-a-m does not solve the issue in my case. Any idea? 

There must be some more keys that are missing .