cantools/cantools #190: building the info manual: tries to fetch images from online

Apteryks, 2020-04-03:
When building the info manual ('make info' from the docs/ subdir), I get the following error (build happens in a container without networking):

```
WARNING: Could not fetch remote image: https://travis-ci.org/eerimoq/cantools.svg?branch=master [HTTPSConnectionPool(host='travis-ci.org', port=443): Max retries exceeded with url: /eerimoq/cantools.svg?branch=master (Caused by NewConnectionError('<urllib3.connection.VerifiedHTTPSConnection object at 0x7ffff44cbd10>: Failed to establish a new connection: [Errno -2] Name or service not known'))]
WARNING: Could not fetch remote image: https://ci.appveyor.com/api/projects/status/github/eerimoq/cantools?svg=true [HTTPSConnectionPool(host='ci.appveyor.com', port=443): Max retries exceeded with url: /api/projects/status/github/eerimoq/cantools?svg=true (Caused by NewConnectionError('<urllib3.connection.VerifiedHTTPSConnection object at 0x7ffff44cb990>: Failed to establish a new connection: [Errno -2] Name or service not known'))]
WARNING: Could not fetch remote image: https://coveralls.io/repos/github/eerimoq/cantools/badge.svg?branch=master [HTTPSConnectionPool(host='coveralls.io', port=443): Max retries exceeded with url: /repos/github/eerimoq/cantools/badge.svg?branch=master (Caused by NewConnectionError('<urllib3.connection.VerifiedHTTPSConnection object at 0x7ffff44cbdd0>: Failed to establish a new connection: [Errno -2] Name or service not known'))]
WARNING: Could not fetch remote image: https://github.com/eerimoq/cantools/raw/master/docs/monitor.png [HTTPSConnectionPool(host='github.com', port=443): Max retries exceeded with url: /eerimoq/cantools/raw/master/docs/monitor.png (Caused by NewConnectionError('<urllib3.connection.VerifiedHTTPSConnection object at 0x7ffff44cb810>: Failed to establish a new connection: [Errno -2] Name or service not known'))]
done
writing... done
copying Texinfo support files... Makefile done
build succeeded, 5 warnings.
```

I don't care much about the appveyor and coveralls badges, but it'd be nice if the 'monitor.png', which exists in the repo, would be used instead of trying to fetch it from online.

eerimoq, 2020-04-03:
Sounds fair. Feel free to create a PR.

eerimoq, 2020-04-03:
Hmm, on second thought, make sure the rendered readme looks good on Github as well. It is probably not possible to use a local repo image, but I'm not sure.

jgarte, 2025-06-22:
@Apteryks Hi, what's the latest status on this issue?