cantools/cantools #193: Unpacking multiplexed data steps on existing struct data (C-code generator)

basketofkittens, 2020-04-05:
First off let me thank the authors and contributors for building such an awesome library. It has helped me immensely in my CAN projects.

Now to my question: I'm using the C-code generator and I'm having some issues with unpacking multiplexed signals. CAN frames are unpacked as they come in and I have a timer that periodically fires to decode the raw values and display them to the user. 

At the opening of each unpack function there's a memset line that zeroes out the struct you pass in. This is wreaking havoc with my display function since the struct only contains valid data every Nth multiplexed frame. I can work around the problem by clearing individual struct members inside of the mux switch, but this doesn't seem like a great solution.

Original (generated) code:
```c++
int link_mux_msg0_unpack(
    struct link_mux_msg0_t *dst_p,
    const uint8_t *src_p,
    size_t size)
{
    if (size < 8u) {
        return (-EINVAL);
    }

    memset(dst_p, 0, sizeof(*dst_p));

    dst_p->mux |= unpack_right_shift_u16(src_p[0], 0u, 0xffu);
    dst_p->mux |= unpack_left_shift_u16(src_p[1], 8u, 0xffu);

    switch (dst_p->mux) {

    case 0:
        dst_p->speed |= unpack_right_shift_u16(src_p[2], 0u, 0xffu);
        dst_p->speed |= unpack_left_shift_u16(src_p[3], 8u, 0xffu);
        dst_p->clt |= unpack_right_shift_u16(src_p[4], 0u, 0xffu);
        dst_p->clt |= unpack_left_shift_u16(src_p[5], 8u, 0xffu);
        dst_p->fuel_p |= unpack_right_shift_u16(src_p[6], 0u, 0xffu);
        dst_p->fuel_p |= unpack_left_shift_u16(src_p[7], 8u, 0xffu);
        break;
```

Modified code:

```c++
int link_mux_msg0_unpack(
    struct link_mux_msg0_t *dst_p,
    const uint8_t *src_p,
    size_t size)
{
    if (size < 8u) {
        return (-EINVAL);
    }

    memset(&dst_p->mux, 0, sizeof(dst_p->mux));

    dst_p->mux |= unpack_right_shift_u16(src_p[0], 0u, 0xffu);
    dst_p->mux |= unpack_left_shift_u16(src_p[1], 8u, 0xffu);

    switch (dst_p->mux) {

    case 0:
        memset(&dst_p->speed, 0, sizeof(dst_p->speed));
        memset(&dst_p->clt, 0, sizeof(dst_p->clt));
        memset(&dst_p->fuel_p, 0, sizeof(dst_p->fuel_p));
        dst_p->speed |= unpack_right_shift_u16(src_p[2], 0u, 0xffu);
        dst_p->speed |= unpack_left_shift_u16(src_p[3], 8u, 0xffu);
        dst_p->clt |= unpack_right_shift_u16(src_p[4], 0u, 0xffu);
        dst_p->clt |= unpack_left_shift_u16(src_p[5], 8u, 0xffu);
        dst_p->fuel_p |= unpack_right_shift_u16(src_p[6], 0u, 0xffu);
        dst_p->fuel_p |= unpack_left_shift_u16(src_p[7], 8u, 0xffu);
        break;
```

I'm mainly trying to understand if this is a problem with the generated code or if I'm just committing a faux pas by naively interacting only with that one raw data struct.

Thanks!

Karl

eerimoq, 2020-04-05:
Sounds like a good suggestion to leave signals unmodified. I think I added the memsets to allow using ``|=`` everywhere when unpacking signals, for slightly simpler code generator logic.

I suggest this:

``` c
int link_mux_msg0_unpack(
    struct link_mux_msg0_t *dst_p,
    const uint8_t *src_p,
    size_t size)
{
    if (size < 8u) {
        return (-EINVAL);
    }

    dst_p->mux = unpack_right_shift_u16(src_p[0], 0u, 0xffu);
    dst_p->mux |= unpack_left_shift_u16(src_p[1], 8u, 0xffu);

    switch (dst_p->mux) {

    case 0:
        dst_p->speed = unpack_right_shift_u16(src_p[2], 0u, 0xffu);
        dst_p->speed |= unpack_left_shift_u16(src_p[3], 8u, 0xffu);
        dst_p->clt = unpack_right_shift_u16(src_p[4], 0u, 0xffu);
        dst_p->clt |= unpack_left_shift_u16(src_p[5], 8u, 0xffu);
        dst_p->fuel_p = unpack_right_shift_u16(src_p[6], 0u, 0xffu);
        dst_p->fuel_p |= unpack_left_shift_u16(src_p[7], 8u, 0xffu);
        break;
```

basketofkittens, 2020-04-05:
@eerimoq thank you for the quick reply! Your suggested modification seems to work perfectly. 

Ditching the memsets and simply overwriting the existing values seems like a fairly straightforward modification to the code generator. While I'm not a particularly skilled programmer, I'm happy to take a crack at making those changes.

Thank you again. Eller så kan vi ta det på svenska. Tack för hjälpen. :)

eerimoq, 2020-04-05:
Så lite så :)

basketofkittens, 2020-04-10:
@eerimoq I've pushed some changes to my own branch of the repo that I think address the problem, but I'm not sure if I should create a pull request yet (I've literally never contributed to a github project before). The tests that compare to the previously generated C code obviously fail now.

https://github.com/basketofkittens/cantools/commit/ae6c8925fd3c68e3df444b339c4ff8f3f565f604

Should I re-generate the C files and include those? I'm not quite sure how to proceed...

eerimoq, 2020-04-10:
Yes, uncomment the line below and execute the tests (`python3 setup.py test`) to re-generate the C-files. If they look good and all tests pass, create a pull request and I'll review it.

https://github.com/eerimoq/cantools/blob/master/tests/test_command_line.py#L42

eerimoq, 2020-04-16:
Any progress?