cantools/cantools #194: message length unrecognized when start with '0'

hdnauth, 2020-04-06:
when candump out as below, (message length is '08')
  can0  29A  [08]  2B 2A 8F 00 00 40 00 00

below command shows nothing.
  candump can0 | cantools decode AA.dbc

after fixing dump message as below, and when I test using cat command, decoded signals shown well. (message length is '8')
  can0  29A  [8]  2B 2A 8F 00 00 40 00 00

hdnauth, 2020-04-06:
maybe below modification would work,

-RE_CANDUMP = re.compile(r'^.*  ([0-9A-F]+)   \[\d+\]\s*([0-9A-F ]*)$')
+RE_CANDUMP = re.compile(r'^.*\s+([0-9A-F]+)\s+\[[0-9]+\]\s*([0-9A-F ]*)$')

eerimoq, 2020-04-06:
The problem is not the `0`, right, but white spaces? Can you show the exact candump output for the failing case?

hdnauth, 2020-04-06:
right, the fixed white space length seems problem.
this is a CAN-FD message dump.

  can0  29A  [08]  F0 43 5D 00 00 40 00 00
  can0  29A  [08]  23 04 5E 00 00 40 00 00
  can0  29A  [08]  2B 2A 5F 00 00 40 00 00
  can0  29A  [08]  85 8B 60 00 00 40 00 00
  can0  29A  [08]  9B 14 61 00 00 40 00 00
  can0  29A  [08]  48 53 62 00 00 40 00 00
  can0  29A  [08]  3D 9B 63 00 00 40 00 00
  can0  29A  [08]  EE DC 64 00 00 40 00 00
  can0  29A  [08]  F6 1B 65 00 00 40 00 00
  can0  29A  [08]  2B 2A 66 00 00 40 00 00
  can0  29A  [08]  50 94 67 00 00 40 00 00
  can0  29A  [08]  83 D3 68 00 00 40 00 00
  can0  29A  [08]  41 0A 69 00 00 40 00 00
  can0  29A  [08]  92 4D 6A 00 00 40 00 00
  can0  29A  [08]  E7 85 6B 00 00 40 00 00

eerimoq, 2020-04-06:
That dump does not have the correct number of white spaces. The output from candump does not have one white space everywhere, right?

Press the "Insert code" button when posting code, to preserve white spaces.

Something like this: `  can0  29A  [08]  92 4D 6A 00 00 40 00 00`

hdnauth, 2020-04-06:
I tried to build using modified decode.py file. but not succeeded, (maybe) because test codes are not modified. How can I get a fixed version?

eerimoq, 2020-04-06:
I can fix it later today, but first, please see my previous comment. I need the exact candump output to implement a test case.

hdnauth, 2020-04-06:
the dump message are from candump.

  can0  29A  [08]  1F 9D C5 00 00 40 00 00
  can0  29A  [08]  CC DA C6 00 00 40 00 00
  can0  29A  [08]  B9 12 C7 00 00 40 00 00

eerimoq, 2020-04-06:
Ohh that's true, the inlice code does not show all whitespaces. I edited your comment to show them.

eerimoq, 2020-04-06:
A fix is part of release 33.2.0, available on PyPI.

hdnauth, 2020-04-07:
Thanks !!
but I'm still in trouble. How can I build new version?
just download the .zip & make is not working on my raspberry pi...

eerimoq, 2020-04-07:
Try installing it with pip.

closed 2020-04-11 by eerimoq