cantools/cantools #198: dbc file - extended id, different source addresses or destination addresses?

jma1123, 2020-04-15:
Howdy, How do you use the database with multiple SA addresses as extended 21 bit identifies as in J1939?  Some messages are broadcast, and others have SA and DA (destination address).  In general, I do not want to create separate dbc messages for each combination.  How would you suggest receiving specific messages?  For example, one system has three of the same message on the bus with different SA & DA.  This seems to imply that I must setup a generic received handling and then add code to examine each message.  However, I still cannot apply it to a dbc message as the id's are different and the data for each variant of the message is different.  Sorry - just looking for suggestions as this seems like an oversight without the ability to set SA / DA in the message.

geo4711, 2020-04-17:
Hi,
J1939 has two reserved values for SA/DA:
0xFF --> Broadcast (only applicable for DA)
0xFE --> "any"
At least, editing DBCs with Vector CANdb++ supports this logic, and as far as I can remember, Vector CANoe recognizes that special value 0xFE. I. e. define a frame with SA = 0xFE, and CANoe will decode any frame matching the PGN (and "ignores" the SA).
But I guess that's moreless a matter of the tool you're using to decode message. cantools supports decoding, but I haven't use this functionality by now (I've only used it for patching DBCs). So I don't know it looks for a matching message with 0xFE is available, if no one with a specific DA/SA is found.
If you're thinking about changing cantools, you should reflect that this "wildcard" method is J1939 specific. So maybe it's not good idea to apply it to any kind of dbc, but for J1939 databases only.

Besides that, please keep in mind that J1939 has two different types of Identifiers ("PDU format" 1 and 2). Only one format supports the DA (in the other format, the "DA bits" have a different meaning).

jma1123, 2020-04-21:
The tools I have work fine also work fine and allow for 0xFE addresses.  The cantool implementation does not allow for this and would require a 1:1 mapping of database entries to messages transmitted and received.  Yes PDU1 vs. PDU2 is a complication with J1939 but this is easily understood and mapped to PGN ranges.  This would not need to be implemented and the user would simply populate the DA with the correct information.  I do not need a J1939 implementation with ISO address claim, transport protocol, etc. as it's a bit overkill for my application.  

With the present implementation of cantools and python-can using a database may actually be more work.  Updating database signals is more of a hassle and the "periodic" message transmit in the broadcast manager is pointless (stop / start can be done, but the better implementation would be to point to the data which is to be periodically transmitted, the data store should be something like the thread-safe deque; allowing new data to be sent without stopping & starting each time the data changes).  Adding all my messages into the database versus an application is about the same work.  In the application at least I can more easily update the data & encode it when sending messages.  Even a few more functions to cantools such as clone and update identifier (RAM) would be a big improvement.

SebastienAuclair, 2021-02-25:
Have the same issues...
Have been trying for hours to understand why a new DBC is incapable of translating messages that always worked before...
Turnsout this is because it contains a bunch of global 0x____FE can-ids... (i.e. 0xPGNfe) which means that it should be accepting any can-id containing 0xPGNxx.... so regardless of the xx source address, it should be capable of decoding the SPNs.
This forces me to try and catch a failure of decode_message(msg.arbitration_id, msg.data)  to force an SA value of FE replacing xx at the end of the arbitration_id....
There should be a small internal call so that right before returning the "Not find" exception, cantools could do a quick check to see if the PGN + source address FE could be in the dbc somewhere ! (That is what CANalyzer is doing internally.)
Could be a nice new feature to add.

jma1123, 2021-02-26:
Howdy,

As a small item of interest, you are applying an SAE J1939 specific scheme to a generic database.  The database supports 11 bit and 29 bit identifiers.  The concept of transmitting nodes, source addresses, etc. are constructs in the higher level protocol.  

You would have to wrap your own routine around cantools to make it work.  Alternatively you could define unique messages which include the source addresses.  The latter is what I have done as a work around.  

If you are going to make a port of the library or new library, please also include an easier way to create a period database message and update the data in the message.  The ‘encode’ process prevents this in the present implementation.

There is a beta python J1939 that you might want to look into…

Cheers - J

SebastienAuclair, 2021-02-26:
> CAUTION: This email originated from outside of the organization.

Thanks for the reply....

We know that the PGN is present in the DBC but the SA will be global in many cases...

Is there an easy way to use .decode_message() which just the PGN ?

Thanks !

This email message (including its attachments) is confidential and may contain privileged information and is intended solely for the use of the individual and/or entity to whom it is addressed. If you are not the intended recipient of this e-mail you may not disseminate, distribute or copy this e-mail (including its attachments), or any part thereof. If this e-mail is received in error, please notify the sender immediately by return e-mail and make sure that this e-mail (including its attachments), and all copies thereof, are immediately deleted from your system. Please further note that when you communicate with us via email or visit our website we process your personal data. See our privacy policy for more information about how we process it: https://www.volvogroup.com/en-en/privacy.html

jma1123, 2021-02-26:
Howdy,
I didn’t see a way to do it.  Message name or complete arbitration id are what looks available.  I looked at the complete ID in my application.  I would prefer to mask and look at PGN also.  If you find a good work around please post.

J -