cantools/cantools #199: KeyError while decoding extended id's

Tbruno25, 2020-06-03:
dbc file:  https://pastebin.com/MVDx9hKf
sample data log: https://pastebin.com/nEqCBAPJ

while attempting to decode using the dbc and data provided I get the following error:

`  File "C:\python-can\venv\lib\site-packages\cantools\database\can\database.py", line 385, in decode_message
    message = self._frame_id_to_message[frame_id_or_name]
KeyError: 270794816

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<stdin>", line 2, in <module>
  File "C:\python-can\venv\lib\site-packages\cantools\database\can\database.py", line 387, in decode_message
    message = self._name_to_message[frame_id_or_name]
KeyError: 270794816`

eerimoq, 2020-06-04:
The frame ids in the DBC file are invalid. They are more more than 11 bits, and at the same time standard.

I guess cantools could be improved to detect invalid frame ids when loading the DBC file.

Tbruno25, 2020-06-04:
@eerimoq thank you for taking a look!

Somewhat unrelated question -- the dbc came from the [opendbc repo](https://github.com/commaai/opendbc) which seems to follow a non standard format. Is there any way you're aware of to reformat these files so that they're compatible with cantools?

eerimoq, 2020-06-04:
Ok, I see. I think you have to manually replace the invalid frame ids with the correct ones.

Anyway, I'll make cantools through an error if the frame id is too long, as it is invalid and not supported by cantools.

Tbruno25, 2020-06-06:
Ahh I see, I assumed extended id's were supported. Thank you for the clarification!

closed 2020-06-06 by Tbruno25

smpshehan, 2021-03-26:
so cantools doesn't support extended can ids? What library can I use to decode messages from extended can ids?

luym11, 2021-06-09:
> so cantools doesn't support extended can ids? What library can I use to decode messages from extended can ids?

Haven't found any so far. Have you found any that works?

juleq, 2021-06-09:
Cantools does and always has supported extended ids. You could read through #301 to get a little context.

Best regards.