cantools/cantools #200: Is it possible to decode single signal rather than full message?

RobertMouncer, 2020-06-09:
Hello,

Just a quick question is it possiblle to decode a single signal rather than a message?
I have a message within my DBC that contains, let's say, 64 signals. I also have a large log file that needs decoding but I'm only interested in one signal.

In short:
I have a message that contains a signal called 'throttle' and a lot of other parameter.
Is it possible to do somethingn like this:

`dbc.decode_message('Throttle', data_bytes)`

Rather than:
`dbc.decode_message(0x11, data_bytes)`
OR
`dbc.decode_message('Message_name_that_contains_throttle', data_bytes)`

eerimoq, 2020-06-09:
Hello!

No, it is not possible to decode a single signal.

I did implement the bitstruct package in C but I've not integrated that into cantools yet. That should speed up the decoding a lot.

RobertMouncer, 2020-06-09:
Okay thank you for your response!

Do you have any plans on implementing bitstruct C into cantools?

eerimoq, 2020-06-09:
It wasn't as easy as I thought, and it will not speed up decoding as much as I want. Maybe twice as fast. Was hoping for more.

RobertMouncer, 2020-06-10:
Twice as fast is still a massive improvement!