cantools/cantools #201: Error: unpack requires at least 64 bits to unpack (got 32)

jingyuan1994, 2020-06-15:
Hi sir,

Thanks for your sharing. I got one error when decode the message.

The background: I try to use cantools to decode '.blf' file which was saved by CANoe on real bus. Everything seems well when I try a demo, but a error: 'unpack requires at least 64 bits to unpack (got 32)' shown when I try the file I need to decode.

I try to solve the error but nothing useful. Here is part of codes:

{
ist = []
soll = []
D1 = []

db = cantools.db.load_file('HCAN.dbc')
can_log = can.BLFReader('20200527.blf')

for msg in can_log:
  D1 = db.decode_message('EM', msg.data)
  ist.append(D1['IstMoment'])
}

I have no idea about this error. Could you help me to check? Thanks a lot.

eerimoq, 2020-06-15:
The CAN message in the BLF-file has 32 bits data, but the DBC-file says the same message has 64. Either the BLF or the DBC is wrong.

jingyuan1994, 2020-06-15:
Hi eerimoq,

After check, I found the problem is the bug from BLFReader, and I ask python-can's author to fix it.

Thank you! Hope you all the best!

Jingyuan

jingyuan1994, 2020-06-15:
OK...

The same error shown again. 

The bug I solved from python-can is that the blf file was recorded from 2 channels and BLFReader cannot read it well. After solve this problem I translate the blf file to csv and get the message which really has 64 bites data. Also, I tried to run this file in CANoe and it worked. I copy part of the message here:

{
Timestamp: 1590555904.589501        ID: 0057    S Rx                DLC:  4    71 0f 00 00                 Channel: 1
Timestamp: 1590555904.591429        ID: 0090    S Rx                DLC:  8    a5 0c ff 1f c4 09 00 00     Channel: 2
}

The ID is the message 'EM' which shown in the code. The DBC has no reason of wrong (it is for real cars). But it is also true that the error tell me the bytes between blf and dbc are different. Is there any possibility that this error happens because of the same reason that these messages are gotten from two channels?

Thanks!

eerimoq, 2020-06-15:
All I can really say is that the data sizes in the BLF and the DBC does not match. A script that reproduces the problem is needed for further investigation.

jingyuan1994, 2020-06-16:
OK, I fix the bug by changing the codes.

Thanks so much!

closed 2020-06-16 by jingyuan1994