cantools/cantools #202: one interesting question

jingyuan1994, 2020-06-16:
Hi eerimoq,

There is one interesting question I found. I succeed to decode the message and get the value from special signals. But for the same message file (blf), combining with the results I decoded and the value which run in CANoe, there are a little bit difference, just a little, for example the data is decoded as 15 but 13 in CANoe . Something interesting is, the trend is totally right, but there are some difference in value, seems like any parameter didn't multiply. Is that the problem from my dbc? Or CANoe has any special algorithm?

If you need me to share code or something else, in mail will be better.

Thanks!

eerimoq, 2020-06-16:
Hi! Feel free to email more information to erik.moqvist@gmail.com. Try to be very specific to make it easy for me to understand the problem and quickly solve it.

jingyuan1994, 2020-06-16:
Hi, I have send you the email by using my work email. Have you found it? If you did receive it, please check the junk mail. 

Thanks again!

eerimoq, 2020-06-16:
Checking tonight.

closed 2020-06-16 by jingyuan1994