cantools/cantools #203: 64 bit CAN signals loose last few bits with double cast

RonRaf, 2020-06-18:
https://github.com/eerimoq/cantools/blob/c6eb73aa030b9e41ba9f2a60751c7f6ddabd45f5/cantools/database/can/c_source.py#L1181

Data is corrupted if the original data is 64 bit binary CAN signal. This usually would go unnoticed with actual physical signals as 64 bit floating point precision is rarely needed.
But if using the decode and encode functions on binary data, the casting from int to double causes loss of a few LSBs. 

If a signal has no scaling or offset, I used int64 instead of double.

Please excuse me for any lack of GitHub etiquette. First post.

eerimoq, 2020-06-18:
Hello,

I don't quite understand what is wrong. Could you give an example, or possibly a fix?

RonRaf, 2020-06-24:
A lot the Encode and Decode C functions that are generated cast between "double" and some int. I assume "double" is used as it "covers all" (although there is a hit in processing times as double is 64 bit) The problem happens when the original data is a 64 bit integer (signed or unsigned). A 64 bit integer has a higher precision than a double precision float.

Basically this:
[image]

or simpler
[image]

The fix I'm implementing uses double only when the scaling or offset from the DBC has a decimal. All other cases get an int64. It really needs to go deeper than that and look at the range as well. For example, if its only whole numbers but the range is larger than max int64, then double must be used. (I don't have cases like that)

eerimoq, 2020-06-24:
Ok, I see. I'm sure more users are interested in using integer types instead of double when scaling signals. I have no plans on implementing this myself, but I'm open to merge any PR that adds valuable functionality.