cantools/cantools #205: Overlapping signals causes ValueError: Short data.

RobertMouncer, 2020-06-22:
I have attached the dbc to recreate this issue at the bottom under details - GitHub won't let me attach it.
It is fairly common for signals to overlap within dbc files. A good example of this is error flagging and fault codes. A byte might be made up of bit flags that will indicate issues while the byte (fault code) can help detemine the broader area at fault. The dbc file will do a better job at explaining it than I can!

The code to reproduce the error:
```
from cantools import database
dbc = database.load_file('example.dbc',strict=False)
dbc.decode_message(287,[0])
```

The above code will cause:

```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Users\User01\AppData\Local\Programs\Python\Python38\lib\site-packages\cantools\database\can\database.py", line 389, in decode_message
    return message.decode(data, decode_choices, scaling)
  File "C:\Users\User01\AppData\Local\Programs\Python\Python38\lib\site-packages\cantools\database\can\message.py", line 789, in decode
    return self._decode(self._codecs, data, decode_choices, scaling)
  File "C:\Users\User01\AppData\Local\Programs\Python\Python38\lib\site-packages\cantools\database\can\message.py", line 748, in _decode
    decoded = decode_data(data,
  File "C:\Users\User01\AppData\Local\Programs\Python\Python38\lib\site-packages\cantools\database\utils.py", line 95, in decode_data
    unpacked.update(formats.little_endian.unpack(bytes(data[::-1])))
ValueError: Short data.
```

**Is there a way around this?**

<details>

VERSION ""

NS_ : 
	NS_DESC_
	CM_
	BA_DEF_
	BA_
	VAL_
	CAT_DEF_
	CAT_
	FILTER
	BA_DEF_DEF_
	EV_DATA_
	ENVVAR_DATA_
	SGTYPE_
	SGTYPE_VAL_
	BA_DEF_SGTYPE_
	BA_SGTYPE_
	SIG_TYPE_REF_
	VAL_TABLE_
	SIG_GROUP_
	SIG_VALTYPE_
	SIGTYPE_VALTYPE_
	BO_TX_BU_
	BA_DEF_REL_
	BA_REL_
	BA_DEF_DEF_REL_
	BU_SG_REL_
	BU_EV_REL_
	BU_BO_REL_
	SG_MUL_VAL_

BS_:

BU_:

BO_ 287 Message01: 1 Vector__XXX
 SG_ Flag1 : 0|1@1+ (1,0) [0|1] "" Vector__XXX
 SG_ Flag2 : 1|1@1+ (1,0) [0|1] "" Vector__XXX
 SG_ Flag3 : 2|1@1+ (1,0) [0|1] "" Vector__XXX
 SG_ Flag4 : 3|1@1+ (1,0) [0|1] "" Vector__XXX
 SG_ Flag5 : 4|1@1+ (1,0) [0|1] "" Vector__XXX
 SG_ Flag6 : 5|1@1+ (1,0) [0|1] "" Vector__XXX
 SG_ Flag7 : 6|1@1+ (1,0) [0|1] "" Vector__XXX
 SG_ Flag8 : 7|1@1+ (1,0) [0|1] "" Vector__XXX
 SG_ Status : 0|8@1+ (1,0) [0|255] "" Vector__XXX

</details>

eerimoq, 2020-06-22:
CANdb++, which is the reference tool, complains about overlapping signals as well. The Status signal is overlapping with the Flags signals.

A multiplexer is required to have overlapping signals, at least that's my understanding.

RobertMouncer, 2020-06-22:
Hmmm a multiplexer is required when you want to reuse a number of bits/bytes to represent different data - for this a multiplexer signal is required. This multiplexer signal represents the current state of the bits/bytes within the message.

But the issue i raised isn't a multiplexer problem. Some automotive product providers like to use overlapping signals - much like what I have shown in the dbc to show error flags and a status code. 

So judging from your answer - it's not possible to decode from the dbc provided using cantools?

eerimoq, 2020-06-22:
Not with the current implementation. I can investigate if it's easy to implement, and if so, likely implement it.

eerimoq, 2020-06-22:
It doesn't look that easy to implement, sorry.

RobertMouncer, 2020-06-23:
Okay, thanks for your help! I'll have to create an adapter for cantools. 

It may be worth adding this feature to your future plans for the project as it is not unusual for automotive manufactorers to have overlapping signals like in the dbc file originally mentioned :)

Thanks again!

DustinMoriarty, 2020-10-27:
When we do things like this at work we tend to put the larger signal in the DBC and then do the bit masking and shifting as a separate step in software outside of the DBC. The idea of showing the same data both ways in the DBC is an interesting one, though I bet it breaks lots of CAN software stack implementations.

RobertMouncer, 2020-10-27:
@DustinMoriarty Luckily we use a lot of internal tools that handle this. 

When it comes to multiplexed signals, thats where it falls over the most!

DustinMoriarty, 2020-10-27:
@RobertMouncer : That is interesting. Since DBC is more reference design than a standard, it is always enlightening to see how different groups interpret it.

mitchdetailed, 2023-04-19:
Bump, this is severely annoying as well.

andlaus, 2023-04-20:
have you tried padding the raw data if it is too short?
```python
dbmsg = dbc.get_message_by_frame_id(287)
payload = bytearray([0])
if len(payload) < dbmsg.length:
  payload.extend([0]*(dbmsg.length - len(payload)))
dbmsg.decode(payload, strict=False)
```

just don't expect the data of these overlapping signals to be correct...

RRMueller, 2024-11-25:
Bump - Quite an annoying bug.

andlaus, 2024-11-25:
you can also try to pass `allow_truncated=True` to `decode()`...

RRMueller, 2024-11-25:
> Quite an old ticket. The way I remember getting around this programmatically -
> 
> 1. Find the overlapping signals
> 2. Save full dbc file as an object
> 3. Remove the overlapping signals from the dbc file but keep the signal you want to decode.
> 4. Decode the data using the adjusted dbc
> 5. Delete modified dbc and restore full dbc.
> 
> It was a PITA but it made it work. This library saved a lot of hassle for me so creating this shitty workaround wasn't a big deal.

I'll give this a try, I was already considering doing something like it. I'm trying to decode basically the entire J1939 dbc, but I will likely not need any multiplexed signals for my usage.

mitchdetailed, 2024-11-25:
allow truncated = true is what i used iirc, but it's been a couple years
since i tried

RobertMouncer, 2024-11-25:
As far as I remember, cantools handles multiplex signals very well. It's overlapping signals in a dbc that it can't do. Just make sure your dbc file is correct and set to multiplex signals.

> I'll give this a try, I was already considering doing something like it. I'm trying to decode basically the entire J1939 dbc, but I will likely not need any multiplexed signals for my usage.

RRMueller, 2024-11-25:
> As far as I remember, cantools handles multiplex signals very well. It's overlapping signals in a dbc that it can't do. Just make sure your dbc file is correct and set to multiplex signals.
> 

Yea I realize now I had a not-so-great database. Once I fixed it cantools worked great. Thanks for the help, everyone.