cantools/cantools #211: Newest version is Incompatible with Python 3.5

superjax, 2020-07-18:
I don't know if this was intentional, but the last release (Remove Python 2 support) actually broke compatibility with all versions of Python prior to 3.6 through the addition of a dependency on `argparse-addons` (which uses f-strings).

eerimoq, 2020-07-18:
Hello!

It was intentional. f-strings are such a nice feature =) Python 3.6 has been around since December 2016 (almost 4 years), so I figured it was time to start using it.

I could remove the f-strings from `argparse_addons`, but I suggest that users update Python to the latest version. After all, f-strings are useful to everyone (among other new features). However, I do understand if users prefer to continue using their old Python version as it is initially easier for them. It's a pity it's not always trivial to update Python. :/

superjax, 2020-07-21:
I agree, f-strings are awesome, and I wish I could just always use an up-to-date python version.

My issue is that I have a bunch of Xenial machines that are pinned to python 3.5.  It took me a little while to figure out why my automated `pip` tooling was broken, since `pip install cantools` on a xenial machine just spews f-string syntax errors coming from `argparse-addons`, and it took a little sleuthing to figure out why `argparse-addons` was being included in the first place.

It might be nice to add a `python_requires='>3.6.0',` to the `setup.py` in `cantools` to make it obvious that `cantools` is the problem.

I'll send a PR.