cantools/cantools #228: Load DBC file with empty signal groups

filipmarling, 2020-11-06:
I have a DBC file with empty signal groups. The DBC file therefore doesn't load using 
cantools.db.load_file(dbc_filename) 

The following error is given:

cantools.database.UnsupportedDatabaseFormatError: DBC: "Invalid syntax at line 22878, column 56: "SIG_GROUP_ 2566907954 DaigFaultStat_VP232_FAS_GW_sg 1 :>>!<<;""

The error explains the problem very well. There should be signal names at  >>!<< but there isn't.
The line looks like this in the DBC:
SIG_GROUP_ 2566907954 DaigFaultStat_VP232_FAS_GW_sg 1 :;

A signal group should according to Vector be defined as:
signal_groups = 'SIG_GROUP_' message_id signal_group_name repetitions ':' { signal_name } ';' ;

I would prefer if the DBC file could be loaded with empty { signal_name } field.

eerimoq, 2020-11-06:
Feel free to implement it.

filipmarling, 2020-11-06:
I don't think I have the ability but I guess I could give it a shot.

eerimoq, 2020-11-06:
Would be much appreciated, thanks!

filipmarling, 2020-11-06:
I have attempted to implement a fix. It is my first contribution on GitHub so please be gentle if I have done something silly.

closed 2020-11-10 by eerimoq