cantools/cantools #232: decode_message MemoryError

Pi57, 2020-11-18:
Hello,

I'm quite new in python, and I'm trying to use cantools, mainly the CAN decode_message feature.

I'm using Python 3.9 and cantools 35.4.0

My issue is that, when I parse a blf file and decode each message, I can see the memory usage increase until I got an error MemoryError and the script stop.

My loop is this one :

>      for file in fileList:
>          decodedData = []
> 
>          # Read BLF
>          print("Loading BLF file from '{}'.".format(inputFilePath + file))
>          log = can.BLFReader(inputFilePath + file)
> 
>          logFull = list(log)
> 
>          # Keep only message in the filter
>          logLight = [msg for msg in logFull if msg.arbitration_id in keepMsg]
> 
>          for msg in logLight:
>              decodedMessage = db.decode_message(msg.arbitration_id, msg.data)

Maybe I don't use the function correctly ? I can get decoded data correctly, but there is this memory issue.
If I remove the line with decode_message, the memory is fine.

Maybe someone have a clue to solve this issue ?

Thanks

eerimoq, 2020-11-18:
You are not using this project. Create an issue here instead: https://github.com/hardbyte/python-can

Pi57, 2020-11-18:
The project you indicate is for the BLFReader I think, I don't have any issue with this feature.

I got the memory issue when I use the decoding function from this project I think ?

eerimoq, 2020-11-18:
Ohh, sorry, you are right. Could you provide the traceback?

Pi57, 2020-11-19:
You can find the traceback here :
```
Traceback (most recent call last):
  File "test.py", line 38, in <module>
  File "test.py", line 38, in <listcomp>
MemoryError
[9952] Failed to execute script test
```

line 38 is :
`logLight = [msg for msg in logFull if msg.arbitration_id in keepMsg]`

and memory is full

eerimoq, 2020-11-19:
Well, that's before cantools is used, right? So the problem is that the log is too big and your machine runs out of memory?

Pi57, 2020-11-19:
Each log files are around 1.4 MB.

If I parse all files without decoding them, I have no issue with the memory

eerimoq, 2020-11-19:
Could you provide an example I can use to reproduce the issue? Makes it easier for me to troubleshoot it.

Pi57, 2020-11-19:
You mean, you want a blf and a dbc file to test it ?

eerimoq, 2020-11-19:
I think I managed to reproduce it myself.

eerimoq, 2020-11-19:
Update the cantools dependency bitstruct to version 8.11.1.  Something similar to `pip install -U bitstruct`. `-U` is important.

Pi57, 2020-11-19:
ok I will try and let you know

Thanks :)

Pi57, 2020-11-19:
After updating bitstruct dependency, memory issue disappear :)

Thanks for your quick help !

closed 2020-11-19 by Pi57