cantools/cantools #244: no access to variables "ProtocolType", "UseGMParameterIDs", "VFrameFormat" and "BusType" variables

kdschlosser, 2020-12-02:
How the variables are defines in a DBC file

`ProtocolType`
```
BA_DEF_  "ProtocolType" STRING ;
BA_DEF_DEF_  "ProtocolType" "GMLAN";
BA_ "ProtocolType" "GMLAN";
```

`UseGMParameterIDs`
```
BA_DEF_  "UseGMParameterIDs" INT 0 0;
BA_DEF_DEF_  "UseGMParameterIDs" 1;
```

`BusType`
```
BA_DEF_  "BusType" STRING ;
BA_DEF_DEF_  "BusType" "";
BA_ "BusType" "CAN";
```

the 3 above variables  should be accessible through a `cantools.database.can.database.Database` instance
the variable below should be accessible through a `cantools.database.can.message.Message` instance

`VFrameFormat`
```
BA_DEF_ BO_  "VFrameFormat" ENUM  "StandardCAN","ExtendedCAN","StandardCAN_FD","ExtendedCAN_FD","J1939PG";
BA_DEF_DEF_  "VFrameFormat" "";

BA_ "VFrameFormat" BO_ {MESSAGE ID} 0;
```
where {MESSAGE ID} is replaced with the id of the message

It is important to have access to these variables when dealing with a GMLAN CAN data system in GM vehicles. This is because of how the arbitration id's are broken down on these systems and hot the message id's are stored in a DBC file for an extended GMLAN id.

received arbitration id on a GMLAN CAN system

0x104F0099

binary of that arbitration id
`1, 0, 0, 0, 0, 0, 1, 0, 0, 1, 1, 1, 1, 0, 0, 0, 0, 0, 0, 0, 0, 1, 0, 0, 1, 1, 0, 0, 1`

with GMLAN the binary data gets broken down like this.
```
priority                  arbitration_id                                ecu_id
1, 0, 0,        0, 0, 0, 1, 0, 0, 1, 1, 1, 1, 0, 0, 0,     0, 0, 0, 0, 0, 1, 0, 0, 1, 1, 0, 0, 1
```

arbitration_id = 0x278
ecu_id = 0x99
priority = 0x4

Now in the world of DBC files the priority and the ecu_id are not important and are not used in the id of a message
so we want to replace those bits with 0,s
```
priority                  arbitration_id                                ecu_id
0, 0, 0,         0, 0, 0, 1, 0, 0, 1, 1, 1, 1, 0, 0, 0,     0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0
```

arbitration_id = 0x4F0000
message id stored in the dbc file would be 0x4F0000

That is what the purpose of "ProtocolType" and "UseGMParameterIDs" is for. It is the mechanism used to identify the DBC
file as a GMLAN DBC file and that GMLAN Extended id's are being used. This would be the trigger to tell an application 
that it needs to modify the arbitration id on received messages to account for this.

I personally feel that this should be handled internally in cantools and if that is what you would like 
done let me know and I can make the needed changes for bitshifting the frame id of the DBC is GMLAN 
and the GMExtended ids are being used.

I do not even know how to go about accessing the variables needed to do this, I would need a little help 
in isolating those.

Or I can handle it in my application code which is fine also bit I would still need access to the above 
mentioned variables.

using the following code in a DBC file triggers a response when using Vector CANdb++

```
BA_DEF_  "ProtocolType" STRING ;
BA_DEF_  "UseGMParameterIDs" INT 0 0;
BA_DEF_DEF_  "ProtocolType" "GMLAN";
BA_DEF_DEF_  "UseGMParameterIDs" 1;
BA_ "ProtocolType" "GMLAN";
```
when I view a message properties sheet inside of CANdb++ I see the following

[image]

and when I go to change the id I get the follow
[image]

you can see in that last image that I am not allowed to alter the priority or the ecu_id.

I have not been able to isolate exactly what "Reserved" is for. I am going to tinker with it some this evening and see if I am able to isolate the bits that make up this number.

kdschlosser, 2020-12-02:
OK I isolated the reserved bits

```
priority                  arbitration_id                      reserved                   ecu_id
1, 0, 0,        0, 0, 0, 1, 0, 0, 1, 1, 1, 1, 0, 0, 0,     0, 0, 0, 0, 0,       1, 0, 0, 1, 1, 0, 0, 1
```

It appears there is an error in CANdb++ with regards to this.

I cannot set an arbitration id of `0x1FFFFFFF`
`1, 1, 1,     1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1,      1, 1, 1, 1, 1,       1, 1, 1, 1, 1, 1, 1, 1`

It throws an error that is not handled by the application

the highest value that can be set in CANdb++ is `0x1FFFFEFF`
`1, 1, 1,     1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1,     1, 1, 1, 1, 0,       1, 1, 1, 1, 1, 1, 1, 1`

which breaks down to
priority = 0x7
arbitration id = 0x1FFF
reserved = 0x1E
ecu_id = 0xFF

eerimoq, 2020-12-02:
"ProtocolType", "UseGMParameterIDs", "VFrameFormat" and "BusType" should be readable in the `database.dbc` and `message.dbc...` attributes.

I don't have any experience from GMLAN CAN systems. I've used J1939, and there is a file called j1939.py in this repository with some helper functions for that format. I guess the best approach would be to integrate j1939 and GMLAN into the database and message classes, but I'm not sure how to do it to make it useful. Possibly better to create new classes. I'm not sure, and I don't have time nor experience to think too much about it. However, I would gladly merge pull requests that adds useful functionality, I just don't have good input on how to design it.

kdschlosser, 2020-12-02:
This is what I get when I print out `cantools.database.can.database.Database.dbc.attribute_definitions`

```
OrderedDict([('UseGMParameterIDs', attribute_definition('UseGMParameterIDs', 1)), ('ProtocolType', attribute_definition('ProtocolType', GMLAN)), ('BusType', attribute_definition('BusType', ))])
```
the above should read
```
OrderedDict([('UseGMParameterIDs', attribute_definition('UseGMParameterIDs', 1)), ('ProtocolType', attribute_definition('ProtocolType', GMLAN)), ('BusType', attribute_definition('BusType', CAN))])
```
```
OrderedDict([('GenSigStartValue', attribute_definition('GenSigStartValue', 0)), ('ProtocolType', attribute_definition('ProtocolType', )), ('VFrameFormat', attribute_definition('VFrameFormat', ))])
```
```
OrderedDict([('GenSigStartValue', attribute_definition('GenSigStartValue', 0)), ('ProtocolType', attribute_definition('ProtocolType', OBD2)), ('VFrameFormat', attribute_definition('VFrameFormat', ["StandardCAN","ExtendedCAN","StandardCAN_FD","ExtendedCAN_FD","J1939PG"]))])
```
and the above should read

the examples above were done at the database level and they do not change at the message level, They should change at the message level

```
OrderedDict([('GenSigStartValue', attribute_definition('GenSigStartValue', 0)), ('ProtocolType', attribute_definition('ProtocolType', OBD2)), ('VFrameFormat', attribute_definition('VFrameFormat', StandardCAN))])
```

eerimoq, 2020-12-02:
I think @geo4711 implemented most of this. What do you think geo4711?

kdschlosser, 2020-12-02:
I was sifting through the code in the file parser for the dbc files and I do not see anything in that that would marry an attribute that has an id or name specified to that object, nor do I see anything that would grab the value from an ENUM definition.

I am unsure as to why BusType remains blank and so does VFrameFormat. without the express ability to know if an attribute is set for a specific object in my case a message it is going to make it impossible to know what has been set where.

kdschlosser, 2020-12-02:
I can confirm 100% that this does not assign the attribute to the message

```
BA_DEF_  "UseGMParameterIDs" INT 0 0;
BA_DEF_DEF_  "UseGMParameterIDs" 1;
BA_ "UseGMParameterIDs" BO_ 2155175936 0;
```
if I understand how this thing works is the first line declares the attribute the second line sets the global default for the attribute and the 3rd line would set the attribute specific to a message that has the id of 2155175936

The global attribute does get set properly but the object specific one a message in this case does not.

geo4711, 2020-12-02:
> I think @geo4711 implemented most of this. What do you think geo4711?

Hmm, the J1939 features have been already available when I joined the project, as well as attribute access for different classes.
Concerning the issue itself (the special handling of CAN ids at GM):
* It seems that you've got a special version of Vector's CANdb++ (or at least the diaglog for CAN message ID is a hidden feature, depending on the existence of "useGMParameterIDs").
* At least, the GM concept of splitting an extended CAN ID seems to be proprietary (OEM specific), not public (like SAE-J1939 or other stuff).
I'll add separate comments to the other posts ...

geo4711, 2020-12-02:
> I can confirm 100% that this does not assign the attribute to the message
>
> ```
> BA_DEF_  "UseGMParameterIDs" INT 0 0;
> BA_DEF_DEF_  "UseGMParameterIDs" 1;
> BA_ "UseGMParameterIDs" BO_ 2155175936 0; [...]

If I'm right, the first two lines define a database attribute (as there's no keyword like "BO_", "BU_", "SG_".
Did you try to insert this:
```
BA_DEF_  BO_ "UseGMParameterIDs" INT 0 0;
BA_DEF_DEF_ BO_ "UseGMParameterIDs" 1;
```
The third line should fine as is.

Maybe I mixed up something. But I think the only way to have an attribute individual per message is defining it as a message attribute, not a database attribute.

geo4711, 2020-12-02:
> It does still store the arbitration id with bit 31 set if using the GM Extended id

The bit 31 is used in DBCs independent from how you use the real CAN-ID (29 bit).
So no matter if you use standard "CANextended", J1939 (which is on top of CANextended), GMLAN or any other protocol that physically uses 29-bit-CAN-IDs: I guess the bit 31 will always be set.

kdschlosser, 2020-12-02:
OK I did locate part of the problem. I am not sure what is actually happening with the use of the ordered dicts.

if the dbc.py file in the \_load\_messages function I changed the get_attributes function to this

```python
    def get_attributes(frame_id_dbc):
        """Get attributes for given message.

        """
        from collections import OrderedDict

        attrib = OrderedDict()

        for k, v in definitions.items():
            attrib[k] = Attribute(v.default_value, v)

        if frame_id_dbc in attributes and 'message' in attributes[frame_id_dbc]:
            for k, v in attributes[frame_id_dbc]['message'].items():
                attrib[k] = v

        return attrib
```

This way if a global parameter is set it gets provided to the message and if the message has any specific attributes set it will get those as well. This is beneficial so that if a global parameter gets changes through a message it is actually only going to change it for that message and not globally. If wanting to change a global attribute it should be done at the database level.

I have not used the saving portion of the library and it may goof that up. Might have to add some form of a check to see if a message attribute differs from a globally set on and if it does then save it as a message specific attribute.

as far as the 0x800000 being marked on the arbitration id is concerned.
I wanted to make sure that the behavior of CABdb++ was the same when using the UseGMParameterIDs attribute. It changes the behavior of how the arbitration id's are entered I didn't know if there were any other changes in behavior.

CANdb++ only allows the middle 13 bits to be provided as the ID, If a dbc file is loaded that has a full id including the reserved, priority and ecu id CANdb++ strips this information out of the id, I do not believe that this is something that cantools should do. It should leave the id's in the dbc file alone. Maybe give a warning? What I elected to do was to check for a message match using the id that has been given. If no match is found then I iterate over the stored messages in the dbc file and I check each one for the UseGMParameterIDs attribute and see if it has been set to a 1. If it has the attribute and it has been set to a 1 i take the supplied frame id and perform this operation on it `frame_id >> 13 & 0x1FFF` and compare that output to the messages frame id. if there is a match then I return the message

```python
    def get_message_by_frame_id(self, frame_id):
        """Find the message object for given frame id `frame_id`.

        """

        if frame_id & self._frame_id_mask in self._frame_id_to_message:
            return self._frame_id_to_message[frame_id & self._frame_id_mask]

        for message in self.messages:
            if (
                'UseGMParameterIDs' in message.dbc.attributes and
                message.dbc.attributes['UseGMParameterIDs'].value == 1
            ):
                # convert to GM  Parameter Id
                id = frame_id >> 13 & 0x1FFF

                if message.frame_id == id:
                    return message

        raise KeyError(frame_id)
```

I also had to update the encode_message and decode_message methods
```python
        try:
            message = self.get_message_by_frame_id(frame_id_or_name)
        except KeyError:
            message = self.get_message_by_name(frame_id_or_name)
```

kdschlosser, 2020-12-02:
There is something off with the ordered dicts or with the attributes/default attributes. If I print them out they appear to have nothing stored but if I access them they do in fact have data that has been stored. 

The other thing is the get_protocol function in the \_load\_messages function returns a None if J1939PG is not the selected item in the J1939PG. This is an incorrect usage of the vocabulary. The protocol is not derived from the VFrameFormat attribute, the protocol is derived from the ProtocolType attribute. the VFrameFormat attribute is exactly that a frame format or it could also be considered a frame protocol in the loosest meaning. It is there to describe the layout of the bits in a frame. where as the ProtocolType is there to describe the communications protocol that is encapsulated inside of the frame data.

So with GMLAN as an example. I would set the ProtocolType attribute to GMLAN. This is the indicator to look for the UseGMParameterIDs attribute.
If I set the ProtocolType to OBD2 then I would know that the first byte in the data portion of the frame is always going to be the number of bytes that are populated in the frame because in OBD2 the DLC of the frame is always going to be 8 the unused data portions would simply be trailing 0's.

I know that it would break API but I do feel that keeping the vocabulary in line is extremely important. The protocol should be kept in place anywhere that uses it but it should be changed to be populated by the ProtocolType attribute instead. There needs to be methods and variables added for frame_format that would hold the information from the VFrameFormat attribute if that attribute has been set in the users DBC file.

does that make sense?? I know it is kind of a tangle with how I have explained it and I am trying to think of a better way to do it. It might be easier to make the code changes and create a PR if only for the sake of being able to see it that way which would make it easier to see what I am talking about.

kdschlosser, 2020-12-02:
ok so here is some updated files

<details>
  <summary>dbc.py (click to expand)</summary>

```python
# Load and dump a CAN database in DBC format.

import re
from collections import OrderedDict as odict
from collections import defaultdict
from decimal import Decimal
from copy import deepcopy

import textparser
from textparser import Sequence
from textparser import choice
from textparser import ZeroOrMore
from textparser import OneOrMore
from textparser import OneOrMoreDict
from textparser import DelimitedList
from textparser import Any
from textparser import AnyUntil
from textparser import tokenize_init
from textparser import Token
from textparser import TokenizeError
from textparser import Optional

from ..attribute_definition import AttributeDefinition
from ..attribute import Attribute
from ..signal import Signal
from ..signal import Decimal as SignalDecimal
from ..signal_group import SignalGroup
from ..message import Message
from ..node import Node
from ..bus import Bus
from ..internal_database import InternalDatabase
from ..environment_variable import EnvironmentVariable

from .utils import num

DBC_FMT = (
    'VERSION "{version}"\r\n'
    '\r\n'
    '\r\n'
    'NS_ : \r\n'
    '\tNS_DESC_\r\n'
    '\tCM_\r\n'
    '\tBA_DEF_\r\n'
    '\tBA_\r\n'
    '\tVAL_\r\n'
    '\tCAT_DEF_\r\n'
    '\tCAT_\r\n'
    '\tFILTER\r\n'
    '\tBA_DEF_DEF_\r\n'
    '\tEV_DATA_\r\n'
    '\tENVVAR_DATA_\r\n'
    '\tSGTYPE_\r\n'
    '\tSGTYPE_VAL_\r\n'
    '\tBA_DEF_SGTYPE_\r\n'
    '\tBA_SGTYPE_\r\n'
    '\tSIG_TYPE_REF_\r\n'
    '\tVAL_TABLE_\r\n'
    '\tSIG_GROUP_\r\n'
    '\tSIG_VALTYPE_\r\n'
    '\tSIGTYPE_VALTYPE_\r\n'
    '\tBO_TX_BU_\r\n'
    '\tBA_DEF_REL_\r\n'
    '\tBA_REL_\r\n'
    '\tBA_DEF_DEF_REL_\r\n'
[... 2072 lines cut ...]
        lines = []

        lines.append("version('{}')".format(self._version))
        lines.append('')

        if self._nodes:
            for node in self._nodes:
                lines.append(repr(node))

            lines.append('')

        for message in self._messages:
            lines.append(repr(message))

            for signal in message.signals:
                lines.append('  ' + repr(signal))

            lines.append('')

        return '\n'.join(lines)
```

</details>

I did make the changes needed to handle the writing of the file as well with respect to the attributes so if an objects attribute value matches the default value for that attribute a BA_ entry will not be written to the file for that object

kdschlosser, 2020-12-02:
> If I'm right, the first two lines define a database attribute (as there's no keyword like "BO_", "BU_", "SG_".

Unfortunately Vector has the DBC file specification under tight lock and key and the only way to know is to reverse engineer it. I am doing this by making changes in the CANdb++ app saving the file and then doing a before and after file comparison to see what has changed. This is what I have put together as far as how the attributes work, I may be incorrect and if I am please correct me. I am all for an education and I do like to know how every aspect of my applications work. I don't like to blindly use external libraries in my applications.

if the attribute deceleration is an INT it would lay out like this

declare attribute | attribute name | attribute type | min | max 
-----------------|-----------------|---------------|-----|------
BA_DEF_  | "UseGMParameterIDs" | INT | 0 | 0

attribute default | attribute name | default value
-----------------|-----------------|--------------
BA_DEF_DEF_ | "UseGMParameterIDs" | 1

set attribute | attribute name | object type | object id/name | value | note
-------------|-----------------|------------|-----------------|------|------
BA_ | "UseGMParameterIDs" |   |   | 1 | no object specified so this would be a database default covering all objects
BA_ | "UseGMParameterIDs" | BO_  | 2155175936 | 0 | BO_ = object type message, and next to it is the id of the message we want to set the attribute on

If the value declaration is an ENUM it would lay out like this

declare attribute | attribute name | attribute type | comma separated values
-----------------|-----------------|---------------|------------
BA_DEF_  | "VFrameFormat" | ENUM | "StandardCAN","ExtendedCAN","StandardCAN_FD","ExtendedCAN_FD","J1939PG"

sets the default to an empty string

attribute default | attribute name | default value
-----------------|-----------------|--------------
BA_DEF_DEF_ | "VFrameFormat" | ""

setting the attribute to specific messages and the value is 0 
set attribute | attribute name | object type | object id/name | value 
-------------|-----------------|------------|-----------------|------
BA_ | "VFrameFormat" | BO_  | 2015 | 0
BA_ | "VFrameFormat" | BO_  | 2015 | 0

the 0 value should translate to "StandardCAN" which is gotten from the attribute deceleration, the index of 0 in the enumeration is "StandardCAN"

kdschlosser, 2020-12-02:
OK here are the changes that should properly handle the GM Extended frame ids, J1939 frame ids and the proper handling of the VFrameFormat, UseGMParameterIDs and ProtocolType variables.

This is going to check if there is a message for the given frame id and if there isn't then it will check the frame format type and if the frame format type is either GM or J1939 it will modify the passed in frame id according to specification and then it will check to see if there is a message that exists with that id, and if there is it will return that messge.

<details>
  <summary>message.py (click to expand)</summary>

```python
# A CAN message.

import binascii
from copy import deepcopy

from ..utils import format_or
from ..utils import start_bit
from ..utils import encode_data
from ..utils import decode_data
from ..utils import create_encode_decode_formats
from ..errors import Error
from ..errors import EncodeError
from ..errors import DecodeError

class Message(object):
    """A CAN message with frame id, comment, signals and other
    information.

    If `strict` is ``True`` an exception is raised if any signals are
    overlapping or if they don't fit in the message.

    """

    def __init__(self,
                 frame_id,
                 name,
                 length,
                 signals,
                 comment=None,
                 senders=None,
                 send_type=None,
                 cycle_time=None,
                 dbc_specifics=None,
                 is_extended_frame=False,
                 bus_name=None,
                 signal_groups=None,
                 strict=True,
                 protocol=None,
                 frame_format=None):
        frame_id_bit_length = frame_id.bit_length()

        if is_extended_frame:
            if frame_id_bit_length > 29:
                raise Error(
                    'Extended frame id 0x{:x} is more than 29 bits in '
                    'message {}.'.format(frame_id, name))
        elif frame_id_bit_length > 11:
            raise Error(
                'Standard frame id 0x{:x} is more than 11 bits in '
                'message {}.'.format(frame_id, name))

        self._frame_id = frame_id
        self._is_extended_frame = is_extended_frame
        self._name = name
        self._length = length
        self._signals = signals
        self._signals.sort(key=start_bit)
        self._comment = comment
        self._senders = senders if senders else []
        self._send_type = send_type
        self._cycle_time = cycle_time
        self._dbc = dbc_specifics
[... 3043 lines cut ...]
    def __repr__(self):
        lines = []

        lines.append("version('{}')".format(self._version))
        lines.append('')

        if self._nodes:
            for node in self._nodes:
                lines.append(repr(node))

            lines.append('')

        for message in self._messages:
            lines.append(repr(message))

            for signal in message.signals:
                lines.append('  ' + repr(signal))

            lines.append('')

        return '\n'.join(lines)

```

</details>

eerimoq, 2020-12-03:
Please create a draft PR to make it easier for me to see what was changed. Thanks =)

kdschlosser, 2020-12-03:
I can do that, give me 10 minutes or so to get it done.

eerimoq, 2020-12-03:
Thanks!

First of all, is GMLAN public? I can't find it in my CANdb++ installation. Is there any configuration missing?

In the PR the frame id is masked based on frame format found in the dbc attribute. This violates the database, message and signal classes design as file format agnostic. The dbc specific attribute should not be accessed in these classes, except as storage of DBC specific information. We could add more attributes to these classes if needed to make frame format masking possible if needed.

There is already a frame_id_mask parameter in the database constructor to ignore given bits. It might not be that convenient, but might be useful in you case.

Could you provide an example script to showcase how you want to use cantools? It would give an even better understanding of which changes are needed.

kdschlosser, 2020-12-03:
I am still in the process of trying to understand  how the damned attributes work.

if I use this code the option to set the message id as a GM Extended id is available
```
BA_DEF_  "UseGMParameterIDs" INT 0 0;
BA_DEF_DEF_  "UseGMParameterIDs" 1;
BA_ "UseGMParameterIDs" 1;
```

if I use this code the option to set the message id as a GM Extended id is available
```
BA_DEF_  "UseGMParameterIDs" INT 0 0;
BA_DEF_DEF_  "UseGMParameterIDs" 0;
BA_ "UseGMParameterIDs" 1;
```

if I use this code the option to set the message id as a GM Extended id is available
```
BA_DEF_  "UseGMParameterIDs" INT 0 0;
BA_DEF_DEF_  "UseGMParameterIDs" 1;
```

if I use this code the option to set the message id as a GM Extended id is NOT available
```
BA_DEF_  "UseGMParameterIDs" INT 0 0;
BA_DEF_DEF_  "UseGMParameterIDs" 1;
BA_ "UseGMParameterIDs" 0;
```

if I use this code the option to set the message id as a GM Extended id is NOT available
```
BA_DEF_  "UseGMParameterIDs" INT 0 0;
BA_DEF_DEF_  "UseGMParameterIDs" 0;
BA_ "UseGMParameterIDs" 0;
```

if I use this code the option to set the message id as a GM Extended id is NOT available
```
BA_DEF_  "UseGMParameterIDs" INT 0 0;
BA_DEF_DEF_  "UseGMParameterIDs" 0;
``'

Now I am not sure as to why I can set the default to 0 and set a network attribute as a 1 and it effects the message

I am not able to do this I get an error when loading the file in CANdb++
```
BA_DEF_  "UseGMParameterIDs" INT 0 0;
BA_DEF_DEF_  "UseGMParameterIDs" 0;
BA_ "UseGMParameterIDs" BO_ 1 1;
```

it appears as tho the last line syntax is selective in what it works on.

geo4711, 2020-12-03:
Well, the syntax for defining and using attributes in a DBC is a little bit complicated.
I'll try to explain it.
1) use BA_DEF to define an attribute:
`BA_DEF_ <object type> "<attribute name>" <type> <min> <max>;`
2) use BA_DEF_DEF to set the default value:
`BA_DEF_DEF "<attribute name>" <default value>;`
BA_DEF_ and BA_DEF_DEF_ are mandatory! It's not allowed to omit one of them.
3) use BA_ (optionally) to set specific values for a certain object:
`BA_ "<attribute name>" <object type> <object identifier> <specific value>;`

Keep in mind that an attribute is always defined for a specific object type:
* BO_ for messages
* SG_ for signals
* BU_ for nodes
* EV_ for environement variables
* '' (empty string/no qualifier) for the database/the network itself (this seems true for "useGMParameterIDs").

I guess the parameter "useGMParameterIDs" lets CANdb++ know how to present the message dialog (to split the 29 bit CAN id into the GM specific logic and let the user change the "GM ID" only). However, this attribute does not have different values for different objects (as it is defined as a databases attribute, not a message attribute).
Applying this logic your examples seem to be fine:
If you want the database attribute "UseGMParameterIDs" to be 1, you can do one of the following:
* set the default value (BA_DEF_DEF_) to 1, and DON'T set a different value with BA_
* set the default value to any value, and set the resulting value to 1

I haven't worked much with database attributes. But maybe they should have one instance only (the dbc itself). For that it might be confusing to set the default value and then set a different value with an extra BA_ line.
The syntax makes more sense for attributes of other types. Let's take messages:
```
BA_DEF_ BO_  "GenMsgCycleTime" INT 0 50000;
BA_DEF_DEF_  "GenMsgCycleTime" 0;
BA_ "GenMsgCycleTime" BO_ 2147483705 1000;
```
--> The message 2147483705 will have cycle time 1000, all others 50000.

For more details, you could ask Vector to give you their DBC format specifications. As it's confidential, I can't upload it. But you should receive it for internal use. The attribute syntax is specified in that document. However, I had many questions with Vector, because the spec lacks many aspects ...

kdschlosser, 2020-12-03:
what I would like to know is what the pre defined attribute names are and which ones work what ways.

It seems that the VFrameFormat really doesn't do anything with respect to adding additional dialogs for J1939 and the frame id. So I am going to guess that there is another attribute like the `UseGMParameterIds` variable except for J1939 something like `UseJ1939ParamaterGroupNumbers` or `UserJ1939PGNs`.

So now my question is also this

What does this do
BA_DEF_ BO_  "GenMsgCycleTime" INT 0 50000;

that is different from this?
BA_DEF_ "GenMsgCycleTime" INT 0 50000;

I am going to assume that it is to set the attribute to messages. and if that is the case then how come this works?
BA_DEF_DEF_  "GenSigStartValue" 0;
BA_ "GenSigStartValue" SG_ 2015 Odometer 0;

You would think it would have to be
BA_DEF_DEF_ SG_  "GenSigStartValue" 0;
BA_ "GenSigStartValue" SG_ 2015 Odometer 0;

then working under that assumption this should work as well (haven't tried it yet)
BA_DEF_ BO_ "UseGMParameterIDs" INT 0 0;
BA_DEF_DEF_  "UseGMParameterIDs" 0;
BA_ "UseGMParameterIDs" BO_ 1 1;

kdschlosser, 2020-12-03:
then if I do this
```
BA_DEF_ "GenMsgCycleTime" INT 0 50000;
BA_DEF_DEF_  "GenMsgCycleTime" 0
```

do all objects in the network have the parameter `GenMsgCycleTime` with a default of 50000 or only the network it's self?

There also does not seem to be a separation in cantools for the variable definitions,
```
BA_DEF_ BO_ "GenMsgCycleTime" INT 0 50000;
```
that should be placed into definitions['messages'] right now all of the definitions are lumped together. if defining an attribute this way means that all messages have the attribute then that attribute should be able to be see in Message.attributes, this does not seem to be the case, or maybe I am missing it somewhere?

kdschlosser, 2020-12-03:
ok so in the tests there is `test_issue_167_long_names_from_scratch`
in the test it uses this code to create a database
```python
db = cantools.database.Database(
            messages=[
                can.message.Message(
                    frame_id=1,
                    name='MSG456789_123456789_123456789_ABC',
                    length=8,
                    signals=[
                        can.signal.Signal(name='SIG456789_123456789_123456789_ABC',
                                          start=9,
                                          length=8)
                    ],
                    senders=['NODE56789_abcdefghi_ABCDEFGHI_XYZ']),
                can.message.Message(
                    frame_id=2,
                    name='MSG_short',
                    length=8,
                    signals=[
                        can.signal.Signal(name='SIG_short', start=1, length=8)
                    ],
                    senders=['NODE_short'])
            ],
            nodes=[
                can.node.Node('NODE_short', None),
                can.node.Node('NODE56789_abcdefghi_ABCDEFGHI_XYZ', None)
            ],
            version='')
```

and this is the database output as a string
```
VERSION ""

NS_ : 
	NS_DESC_
	CM_
	BA_DEF_
	BA_
	VAL_
	CAT_DEF_
	CAT_
	FILTER
	BA_DEF_DEF_
	EV_DATA_
	ENVVAR_DATA_
	SGTYPE_
	SGTYPE_VAL_
	BA_DEF_SGTYPE_
	BA_SGTYPE_
	SIG_TYPE_REF_
	VAL_TABLE_
	SIG_GROUP_
	SIG_VALTYPE_
	SIGTYPE_VALTYPE_
	BO_TX_BU_
	BA_DEF_REL_
	BA_REL_
	BA_DEF_DEF_REL_
	BU_SG_REL_
	BU_EV_REL_
	BU_BO_REL_
	SG_MUL_VAL_

BS_:

BU_: NODE_short NODE56789_abcdefghi_ABCDEFGHI_XY

BO_ 1 MSG456789_123456789_123456789_AB: 8 NODE56789_abcdefghi_ABCDEFGHI_XY
 SG_ SIG456789_123456789_123456789_AB : 9|8@1+ (1,0) [0|0] "" Vector__XXX

BO_ 2 MSG_short: 8 NODE_short
 SG_ SIG_short : 1|8@1+ (1,0) [0|0] "" Vector__XXX

BA_DEF_ BU_  "SystemNodeLongSymbol" STRING ;
BA_DEF_ BO_  "SystemMessageLongSymbol" STRING ;
BA_DEF_ SG_  "SystemSignalLongSymbol" STRING ;
BA_DEF_DEF_  "SystemNodeLongSymbol" "";
BA_DEF_DEF_  "SystemMessageLongSymbol" "";
BA_DEF_DEF_  "SystemSignalLongSymbol" "";
BA_ "SystemNodeLongSymbol" BU_ NODE56789_abcdefghi_ABCDEFGHI_XY "NODE56789_abcdefghi_ABCDEFGHI_XYZ";
BA_ "SystemMessageLongSymbol" BO_ 1 "MSG456789_123456789_123456789_ABC";
BA_ "SystemSignalLongSymbol" SG_ 1 SIG456789_123456789_123456789_AB "SIG456789_123456789_123456789_ABC";

```

Now based on what we have discussed above that would mean that the message `MSG_short` has a name of ""
and that message `MSG456789_123456789_123456789_AB` has a name of "MSG456789_123456789_123456789_ABC"

This would be because the attribute `SystemMessageLongSymbol` has been applied to all messages using the `BO_` identifier and a default value of "" was specified for that attribute and there is no specific setting for the attribute for message 2 (`MSG_short`) so it is going to use the default value of ""

if this is not the case then Vectors API really isn't tickling my pickle if ya get my meaning. It seems like when the  API was added to and consideration for how the API was initially supposed to work was tossed right out the window.

geo4711, 2020-12-04:
> what I would like to know is what the pre defined attribute names are and which ones work what ways.
Unfortunately that is not described in the DBC specification document from Vector.

> It seems that the VFrameFormat really doesn't do anything with respect to adding additional dialogs for J1939 and the frame id. So I am going to guess that there is another attribute like the `UseGMParameterIds` variable except for J1939 something like `UseJ1939ParamaterGroupNumbers` or `UseJ1939PGNs`.
The VFrameFormat J1939 is also evaluated by other tools. Knowing that a Frame is defined as PG (and not as Standard or Extended CAN Frame), these tools do the transport protocol stuff for you. That means you can use these PGs (even with a "DLC" > 8) as it would be a single message. And then just send it. The tool care's about the protocol stuff, i. e. it creates a sequence of different frames (CAN extended frames with each 8 data bytes). You don't have to care about that yourself. Same for receiving PGs > 8 databytes.
 
> So now my question is also this
> 
> What does this do
> BA_DEF_ BO_ "GenMsgCycleTime" INT 0 50000;
--> this defines an attribute called `GenMsgCycleTime` for message objects; type = INT, valid range from 0 to 50000. As far as I know this is one of the (undocumented) J1939 specific attributes. I. e. using a J1939 database, this attribute is used for specifying message cycle times (and not the built-in cycle time attribute).

> that is different from this?
> BA_DEF_ "GenMsgCycleTime" INT 0 50000;
--> yes, that is different! This line omits the "BO_" qualifier and would therefor define a cycle time attribute for the whole database.

> I am going to assume that it is to set the attribute to messages. and if that is the case then how come this works?
> BA_DEF_DEF_ "GenSigStartValue" 0;
> BA_ "GenSigStartValue" SG_ 2015 Odometer 0;
--> This is a different attribut (start value, not cycle time). The two lines list the default value and a specific value for the signal "Odometer" in the message with ID 2015. There should by a third line defining the attribute itself (e. g. `BA_DEF_ SG_ "GenSigStartValue" INT 0 65535;`. If such a line does not exist, I expect the DBC to be invalid.

> You would think it would have to be
> BA_DEF_DEF_ SG_ "GenSigStartValue" 0;
> BA_ "GenSigStartValue" SG_ 2015 Odometer 0;
--> not exactly. The point is, that `BA_DEF_` needs the object qualifier ("SG_" in this case), but `BA_DEF_DEF_` does not even allwo it. The reference is just made by the name of the attribute.

> then working under that assumption this should work as well (haven't tried it yet)
> BA_DEF_ BO_ "UseGMParameterIDs" INT 0 0;
> BA_DEF_DEF_ "UseGMParameterIDs" 0;
> BA_ "UseGMParameterIDs" BO_ 1 1;
--> Yes, that looks correct (I'm not sure about `INT 0 0`. This could mean there's no explicit definition of min/max.
But one question:
Is "UseGMParameterIDs" something you want to define per message? I. e. you have some messages in GM style and others in a non GM style? If yes, that snippet above makes sense

geo4711, 2020-12-04:
> then if I do this
> 
> ```
> BA_DEF_ "GenMsgCycleTime" INT 0 50000;
> BA_DEF_DEF_  "GenMsgCycleTime" 0
> ```
> 
> do all objects in the network have the parameter `GenMsgCycleTime` with a default of 50000 or only the network it's self?

No. The `BA_DEF_` does not have a type qualifier. That means it is an attribute on database level. It cannot be used for messages/signals/nodes/envvars!
Defining a "Database attribute" does not mean to set the default value for all objects in this database!!! The database itself is the (single) object of this type. If you want to set default values for all messages, use `BO_` in the `BA_DEF_` statement for example.

> There also does not seem to be a separation in cantools for the variable definitions,
>
> ```
> BA_DEF_ BO_ "GenMsgCycleTime" INT 0 50000;
> ```
>
> that should be placed into definitions['messages'] right [...]

Yeah, that's maybe a problem. The concept of the DBC file format is to have two line for defining the attribute (and its default value), and then one extra line for every object that does not use the default value.
When reading such a file with cantools, this results in "empty" attribute lists for those objects that use the default values.

geo4711, 2020-12-04:
> BU_: NODE_short NODE56789_abcdefghi_ABCDEFGHI_XY
>
>
> BO_ 1 MSG456789_123456789_123456789_AB: 8 NODE56789_abcdefghi_ABCDEFGHI_XY
>  SG_ SIG456789_123456789_123456789_AB : 9|8@1+ (1,0) [0|0] "" [...]

Yes and no :-)
DBCs can treat two names of an object. A short name (up to 32 chars, mandatory), and a long name (afaik up to 128 chars, optionally).
The snippet above just says, that message 2 has an empty long name. But of course the short name still exists: `MSG_short`.
The short name is a kind of compatibility mode. I guess it's neccessary to support some old tools which are limited to 32 chars per object. However, any up to date tool should support the long names. At first, they read the short name. Then they check, if there's a attribute for the long name. If yes, the present the long name to the user, otherwise the short name. This is completely done in the background (also if you change the name from a short to a long string and save the DBC again ...).
This is one of the 'exceptions' that are in fact defined in the DBC specifications in detail.

I hope, this helps :-)

kdschlosser, 2020-12-04:
OK so when talking about the long names.
I have modified the cantools code so it properly populates the various objects with the default attributes. which in the case of the long name example above cantools is not handling the empty default attribute properly. I have to track down where that problem is. I wanted to confirm my understand to make sure that I do the right thing with respect to that portion of it.

Now with respect to the defining of an attribute, I am going to use the `GenSigStartValue` as an example.
In my project I have a DBC file. in that DBC file are all of the bits needed to decode an incoming OBDII CAN message. decoding a message does not need to make use of `GenSigStartValue`. but to send a message it does. I wanted to make everything work nice and easy so using a multiplexed setup along with `GenSigStartValue` I am able to have the the frame data constructed without needing to know anything  other then the name of the message and the name of the signal. by using the `GenSigStartValue` it sets the default of 0 for all signals including the signals that are only used for decoding. Is there a way to unset an attribute for a specific signal??

kdschlosser, 2020-12-04:
I just did a commit and push to #245 It has address all of the above mentioned topics we discussed, I have also updated the tests and all tests run successfully with the changes I have made.

The attributes are now set into their respective objects correctly and get output when calling as_dbc_str if the attributes value differs from the default set for that object type.

The attribute output of as_dbc_string is now more organized.

closed 2020-12-07 by kdschlosser