cantools/cantools #247: make test: command errors on Windows

dzid26, 2020-12-13:
https://github.com/eerimoq/cantools/blob/34fb50c3e45312279e2e928fc68da890d8ace6f8/Makefile#L151
When running `make test` on WIndows, this seems to produce malformed gcc calls. 
```
make test-c-src
make[2]: Entering directory '/c/backup/cantools'
for f in tests/files/c_source/motohawk.c tests/files/c_source/padding_bit_order.c tests/files/c_source/vehicle.c tests/files/c_source/multiplex.c tests/files/c_source/multiplex_2.c tests/files/c_source/floating_point.c tests/files/c_source/no_signals.c tests/files/c_source/choices.c tests/files/c_source/signed.c tests/files/c_source/my_database_name.c tests/files/c_source/min_max_only_6_0.c tests/files/c_source/abs.c ; do \
    C:\msys64\mingw64\bin\gcc.exe -Wall -Wextra -Wdouble-promotion -Wfloat-equal -Wformat=2 -Wshadow -Werror -Wconversion -Wpedantic -std=c99 -O3 -c $f ; \
done
/bin/sh: line 1: C:msys64mingw64bingcc.exe: command not found
/bin/sh: line 1: C:msys64mingw64bingcc.exe: command not found
/bin/sh: line 1: C:msys64mingw64bingcc.exe: command not found
/bin/sh: line 1: C:msys64mingw64bingcc.exe: command not found
/bin/sh: line 1: C:msys64mingw64bingcc.exe: command not found
/bin/sh: line 1: C:msys64mingw64bingcc.exe: command not found
/bin/sh: line 1: C:msys64mingw64bingcc.exe: command not found
/bin/sh: line 1: C:msys64mingw64bingcc.exe: command not found
/bin/sh: line 1: C:msys64mingw64bingcc.exe: command not found
/bin/sh: line 1: C:msys64mingw64bingcc.exe: command not found
/bin/sh: line 1: C:msys64mingw64bingcc.exe: command not found
/bin/sh: line 1: C:msys64mingw64bingcc.exe: command not found
make[2]: *** [Makefile:151: test-c-src] Error 127
make[2]: Leaving directory '/c/backup/cantools'
make[1]: *** [Makefile:160: test-c] Error 2
make[1]: Leaving directory '/c/backup/cantools'
make: *** [Makefile:147: test] Error 2

C:\backup\cantools>
```
There is something wrong with the slashes.

eerimoq, 2020-12-13:
I'm surprised you got that far. I've only used make on my Ubuntu machine. Will try running the tests in Cygwin.

eerimoq, 2020-12-13:
`make test-c-src` works in Cygwin.

dzid26, 2020-12-14:
Ok, so this `/bin/sh: line 1: C:msys64mingw64bingcc.exe: command not found` was related to enviromental variable I had defined for other project:
[image]
When I changed it to forward slashed `C:/msys64/mingw64/bin/gcc.exe`, or removed completly `make test-c-src`went through.

dzid26, 2020-12-14:
Now it stop at https://github.com/eerimoq/cantools/blob/34fb50c3e45312279e2e928fc68da890d8ace6f8/tests/test.mk#L88
```
make[2]: Leaving directory '/cygdrive/c/backup/cantools'
make -C tests
make[2]: Entering directory '/cygdrive/c/backup/cantools/tests'
test.mk:88: *** multiple target patterns.  Stop.
make[2]: Leaving directory '/cygdrive/c/backup/cantools/tests'
make[1]: *** [Makefile:161: test-c] Error 2
make[1]: Leaving directory '/cygdrive/c/backup/cantools'
make: *** [Makefile:147: test] Error 2
```
Seems like it is caused by a colon in line 9 https://github.com/eerimoq/cantools/blob/34fb50c3e45312279e2e928fc68da890d8ace6f8/tests/test.mk#L9
I am not usre what is a solution https://stackoverflow.com/questions/2100448/multiple-target-patterns-makefile-error/3711394#3711394

eerimoq, 2020-12-14:
If you figure it out you can create a PR =)