cantools/cantools #249: Muliplexed signals in generated C code, use unions

pdietl, 2020-12-19:
What do you think about me making a contribution to the C code generation that uses unions to shrink the struct size in the case of multiplexed signals?

eerimoq, 2020-12-19:
Give it a try. There was some reason for not using unions, but I can't remember what it was.

juleq, 2020-12-19:
I think the reason was MISRA:

https://rules.sonarsource.com/c/RSPEC-953

Probably nice to implement that as an option as arguing against MISRA comes at a considerable effort.

pdietl, 2020-12-19:
Yeah that’s a good point. My use case is a message with forty-something multiplexed signals, some with multiple members. In a small microcontrollers I’m concerned with the extra size of the struct.

juleq, 2020-12-19:
Yes, while the rationale behind that rule is perfectly understandable to me, packing/unpacking with unions seems to be a popular exception for good reasons: It can save on source to maintenance, code size and RAM usage in embedded systems. But it also can add prose and probably additional safety measures to a safety case.

I think being able to produce both styles is valuable, because I could understand people being put off by the verbosity of compliant code.

guy-radford-sunswap, 2023-05-12:
Unfortunately without a fix, multiplexed signals stop the generated code from compiling.

andlaus, 2023-05-12:
> Unfortunately without a fix, multiplexed signals stop the generated code from compiling.

btw: is this a statement about the current master version or this pull request?

andlaus, 2023-05-16:
I'm afraid that you will have to elaborate a bit on this: For me, it seems to work:
```
> cantools generate_c_source tests/files/dbc/multiplex.dbc 
Successfully generated ./multiplex.h and ./multiplex.c.
> gcc -Wall -c multiplex.c
```
Also, please open a new issue if your problem is unrelated to this one...

guy-radford-sunswap, 2023-05-16:
have created a new issue as there are two separate issues, see #566

martonmiklos, 2024-05-22:
Hi folks,

I will need to generated headers from DBC file to a resource constrained PIC32MX MCU, so I would also need unionize the signals in the messages similar to @pdietl .

To keep the flash resource constrains low my plan is to simply cast the generated structs to the TX/RX CAN data fields, so technically I only need a header to get generated.

Do you think it would worth include such a capability upstreamed?

I know that it brings many constrains like:
- The signals endianess not mathcing the target endianess
- Signals could not overlap on non 8 bit boundaries
- Padding needs to be generated
- Non n*8 bit fields declaration might be compiler dependant
but I think it could be still a valid use case when working on resource constrained targets.