cantools/cantools #250: Parsing CDD File

JosephGeorges, 2020-12-22:
Hello, 

I am having some difficulty parsing a CDD file with cantools library. I have the cdd file in the same directory as my python script, so I just pass in the file name. My code looks like this:

```
filename = 'example.cdd' 
cdd_file = cantools.db.load_file(filename)
example_did = cdd_file.get_did_by_name('Application Software Identification')
print(example_did)
```

I want to print out the DID information by giving the DID itself or the did name. But on execution I receive a:

eerimoq, 2020-12-22:
The CDD likely ha lots of bugs s it is barely used. Unfortunately no one is fixing the bugs as far as I know. Pull requests are very welcome.

oomzay, 2021-03-03:
I have rather replicated this issue with https://github.com/eerimoq/cantools/issues/290
I am currently experimenting with solutions that enhance, but don't break the current interface.

oomzay, 2021-03-03:
Closing as duplicated/superceded by https://github.com/eerimoq/cantools/issues/290

closed 2021-03-03 by oomzay