cantools/cantools #253: Multiple DBCs for Multiple can buses

xR3b0rn, 2021-01-13:
It is possible to dump CAN data from all CAN buses available via `candump any`, however, when piping the output of `candump` into `cantools`, it's only possible to provide one single DBC. Besides that, a DBC is not describing more than one network, this may lead to problems, when there are more than one message with the same ID on differen networks. To fix this, I propose the introduction of a command line parameter, which maps the name of the interface to a DBC, e.g. `candump any | cantools decode --bus=vcan0:net0.dbc --bus=vcan1:net1.dbc`.

eerimoq, 2021-01-13:
Try using GNU Parallel. I had another program called `parallel` installed that might have worked, but installed GNU Parallel with `sudo apt-get install -y parallel`. Then run `parallel --line-buffer ::: "./1.sh foo" "./1.sh bar"`. I guess this is only useful if line based decoding is used. Feel free to implement something along the lines you suggested.

xR3b0rn, 2021-01-13:
May I ask, why you made the decision to not support a DBC per bus?

eerimoq, 2021-01-13:
How would I know what people need? =)