cantools/cantools #254: Missing command line tool

rdlaner, 2021-01-14:
I recently went to use the cantools command line tool after having not used it for some time and found that it now generated the following exception:

```
% cantools
Traceback (most recent call last):
  File "/usr/local/bin/cantools", line 5, in <module>
    from cantools.__init__ import _main
  File "/usr/local/lib/python2.7/dist-packages/cantools/__init__.py", line 10, in <module>
    from . import database as db
  File "/usr/local/lib/python2.7/dist-packages/cantools/database/__init__.py", line 9, in <module>
    import diskcache
  File "/usr/local/lib/python2.7/dist-packages/diskcache/__init__.py", line 9, in <module>
    from .core import (
  File "/usr/local/lib/python2.7/dist-packages/diskcache/core.py", line 434
    raise ValueError('disk must subclass diskcache.Disk') from None
                                                             ^
SyntaxError: invalid syntax
```

I assumed this was due to the drop in support for python2.7. I therefore did the following:
`pip uninstall cantools`
`pip3 install cantools`

Both operations completed successfully, however, now I no longer have the command line executable. If I look at /usr/local/bin, I no longer see a cantools executable.

Any idea on how I can get it back?

I'm running Ubuntu 18 and using python3.6.

eerimoq, 2021-01-14:
Installing with `python3 -m pip install -U cantools` and then `python3 -m cantools` should work as a workaround. I can't explain why the executable is gone. Maybe installed in `~/.local/bin/`?

rdlaner, 2021-01-14:
Ah, good call. It is installed in ~/.local/bin/. I always forget that pip can install in different locations for different reasons.

Thanks for the quick response!

closed 2021-01-14 by rdlaner