cantools/cantools #256: Vector IEEE Double (IEEE 754 64bit) value support

nexulm, 2021-01-25:
Are IEEE 754 values correctly working in latest updates for decode_message while using a Vector DBC file?

Input( 64bit Intel IEEE Double): 
\x00\x00\x00\xA0\x29\x75\x49\x40

Code line:
db_can4 = cantools.db.load_file('<filename>.dbc')
print(db_can4.decode_message(int(CAN_MsgId,16), bytes(CANtools_rawdata, 'utf-8')))

Output (Faulty one):
3492642873067199527

Code line as current workaround:
struct.unpack('<d', bytes.fromhex(CAN_rawdata)

Output (Correct value):
50.91532897949219

Which would be a better solution to decode 64bit and 32bit IEEE double/single values with decode_message directly without additional lines/code?

eerimoq, 2021-01-25:
Did they work before?

nexulm, 2021-01-25:
Never used them before in Python!
Is it the right place to ask here or should I put the question to any other place, forum, dev-group, ...?

eerimoq, 2021-01-25:
This is the best place to report bugs and ask questions.

Could you provide a minimal example to reproduce the problem? It would be very helpful.

juleq, 2021-01-25:
A few days ago I was programming some comms between two controllers, one 16bit, one 32bit, both LE machines. I stumbled over the following: IEEE754 does not define word order. The 32bit machine (ARM7) stored its two words of a double swapped. Maybe word order is worth checking here?

nexulm, 2021-01-26:
Please find a sample code + dbc + log for further analysis attached:
[debugging.zip](https://github.com/eerimoq/cantools/files/5872229/debugging.zip)

eerimoq, 2021-01-26:
The frame payload given to `decode_message()` is wrong. `bytes(CANtools_rawdata, 'utf-8')` is `b"'\\x00\\x00\\x00\\xE0\\x28\\x75\\x49\\x40'"`, and should be `b'\x00\x00\x00\xE0\x28\x75\x49\x40'`. Please fix your parsing and convertion to bytes.

nexulm, 2021-01-26:
Sorry, it's still not working for me. I've play around some more hours but also exchanging with bytes.fromhex(...) doesn't work as expected. So I'm still looking for more experienced Python users and some help/guidance here:
`            try:
#                print("IEEE double, decode_message: ", CAN_MsgId, " IN: ", CANtools_rawdata, "OUT: ", 
#                      db_can1.decode_message(int(CAN_MsgId,16), bytes(CANtools_rawdata, 'utf-8')))
                print("IEEE double, decode_message: ", CAN_MsgId, " IN: ", CAN_rawdata, bytes.fromhex(CAN_rawdata), "OUT: ", 
                      db_can1.decode_message(int(CAN_MsgId,16), bytes.fromhex(CAN_rawdata)))
                print("IEEE double, struct.unpack from CAN rawdata: ", CAN_MsgId," IN: ", 
                      CANtools_rawdata, "OUT: ", struct.unpack('<d', bytes.fromhex(CAN_rawdata)))`
[image]

eerimoq, 2021-01-26:
Here you go:
```python
import cantools
import can

db_can1 = cantools.db.load_file('IEEE754.dbc')

for message in can.io.ASCReader('IEEE754_Sampledata.asc'):
    try:
        print(db_can1.decode_message(message.arbitration_id, message.data))
    except:
        pass
```
Output:
```
{'GPS_Latitude_Minutes': 52.94189453125}
{'GPS_Longitude_IEEE_Double': 6.836982250213623}
{'GPS_Latitude_IEEE_Double': 50.91530990600586}
{'GPS_Latitude_Minutes': 52.94189453125}
...
```

nexulm, 2021-01-26:
Great, thnks for this code snippet, much easier and shorter but I get this output:
[image]
?????

eerimoq, 2021-01-26:
weird

nexulm, 2021-01-26:
It's working on the same machine in different manner
1) Windows 10 Anaconda Environment with Python 3.8.3 (**not working**) see screenshot above
2) Virtualbox (Linux Ubuntu 20.04 Guest with Pyhton 3.8.5) => **working** ;-)
[image]

eerimoq, 2021-01-27:
Which version of cantools and bitstruct do you have on both systems?

nexulm, 2021-01-27:
bitstruct v8.11.1 (on both systems)
cantools v20.6.0 (**not** working one, Windows10, Anaconda)
cantools v36.1.0 (working one, Linux Ubuntu)

Anaconda update doesn't solve/update the outdated versions.
"pip install cantools" still confirms 20.6.0 as latest (?)
"pip install cantools=36.1.0" solves the issue for me

BUT:
Why was Anaconda (new installation only a few weeks ago) using such an old cantools version?

Finally:
Can be set as closed!

closed 2021-01-27 by nexulm

eerimoq, 2021-01-27:
I have absolutely no idea who maintains Anaconda. In my opinion it is just trouble using that.