cantools/cantools #258: 36.1.0 version: pickling of database fails with error reported in #204

DevD-official, 2021-01-28:
Dear @eerimoq 

Post version 34.0.0, pickling of database object is failing.
As suggested in #204 bitstruct package has been upgraded to the latest version.
But issue still persists.

Code
------
Below code snippet is shared just for reproducing the issue.

```python
db = cantools.db.load_file(filename=<FILE_NAME>)
f = io.BytesIO()
pickle.dump(db, f)
```

Exception
-----------
TypeError: can't pickle bitstruct.c.CompiledFormatDict objects

eerimoq, 2021-01-28:
In #204 a deepcopy was done, not pickling, so that explains why you have the problem. Seems like the C extension classes have to implement the `__reduce__()` method. I've never done so, have you? Could you try to implement it, I'm kinda busy? =)

DevD-official, 2021-01-28:
Thanks for response.
I have also never implemented ´__reduce__()´ method.
Does this method needs to be implemented for CompiledFormatDict  class in bitstruct ?

eerimoq, 2021-01-28:
I think so, but I'm not sure.

DevD-official, 2021-01-28:
Thanks.
I will close the issue.

closed 2021-01-28 by DevD-official