cantools/cantools #259: Forced installation of matplotlib

AxelVoitier, 2021-01-28:
Hello,

I noticed the new plot feature is creating a hard dependency on matplotlib.
Can this be avoided?

We are installing this package on restricted embedded targets. Just declaring the dependency creates all kind of issues just by installing cantools: from disk usage to compilation time (as it goes through qemu ARM emulation). Especially since matplotlib also depends on numpy...

Can that be declared as an extras in setup.py? And have the main be clever enough to skip on adding subparsers when their importation fails?

Also related is that comment in the main about curses (which actually used to create us issues in the past, as we did not have it on the target :D).

I can send a PR for that later (some times in February, maybe).

Cheers,
Axel

eerimoq, 2021-01-28:
I understand. Feel free to submit a PR that makes all troublesome dependencies optional.

tjhowse, 2021-01-28:
This also affects us: we're running cantools on a resource- and bandwidth-constrained platform and would rather remove the install hacks that uninstall matplotlib after installing cantools :)

segaura, 2021-02-16:
I am running `cantools 36.2.0` on "raspbian-server" without graphical (X) environment, and installed `matplotlib 3.3.4`

```
$ pip show matplotlib
Name: matplotlib
Version: 3.3.4
Summary: Python plotting package
Home-page: https://matplotlib.org
Author: John D. Hunter, Michael Droettboom
Author-email: matplotlib-users@python.org
License: PSF
Location: /home/pi/.local/lib/python3.7/site-packages
Requires: python-dateutil, pyparsing, pillow, kiwisolver, cycler, numpy
Required-by:
```

commands like `cantools decode` shows this error

```
matplotlib package not installed. Required for producing plots.
Traceback (most recent call last):
  File "/home/pi/.local/bin/cantools", line 10, in <module>
    sys.exit(_main())
  File "/home/pi/.local/lib/python3.7/site-packages/cantools/__init__.py", line 38, in _main
    from .subparsers import plot
  File "/home/pi/.local/lib/python3.7/site-packages/cantools/subparsers/plot.py", line 59, in <module>
    plt.rcParams["date.autoformatter.hour"] = "%H:%M"
NameError: name 'plt' is not defined
```

trying to use `matplotlib` alone just to test its installation it doesn't work.

As a workaround I reverted to `cantools 35.5.0` but if there's a way to use the latest version without the need for a _working matplotlib_, it would be better.

AxelVoitier, 2021-02-17:
I will give it a go, I have more time to work on it right now.

AxelVoitier, 2021-02-17:
Actually, #265 should fixes your issue. It's a matter of having it in a released version now.

I will just add an extra in setup.py for plot that will declare a dependency on matplotlib, and document it in README.rst.

eerimoq, 2021-02-17:
@AxelVoitier Feel free to review, merge and tag the fix. Tagged commits are automatically published on PyPI, so it should be easy. I'll be less active in the project from now on and I hope everyone interested in cantools will figure out a way for it to continue being successful.